espressif/esp-idf #1041: GATTC API change

rojer, 2017-09-26:
https://github.com/espressif/esp-idf/commit/1759a47060e4708e264e11b395a0fbaeeca7863a made GATTC API - already quite tricky - even more difficult to use by introducing handles.
previously, i was able to specify service and characteristic UUID directly to `esp_ble_gattc_write_char`, now i need a "handle" for it, which, judging by example, can be obtained only as a result of search and then using some service start and end handle. this is insanity. why was this change made? can we go back to simpler API? can we have the previous, simpler, APIs as alternative to the new ones?

if no, then i have a few questions regarding the use and validity of handles.
 - is performing attribute search mandatory to be able to call `esp_ble_gattc_get_char_by_uuid`?
 - once obtained, how long does the handle remains valid? api docs mention cache, when does it get cleared?

lucazader, 2017-09-26:
I'm not sure if this is in the bluetooth spec, or anything. But as far as I have experienced, the handles are always the same.
Eg say a device has one service with 3 characteristics all will read/write/notify along with a client configuration. The service will have a handle, say 30 in this example. Then handle 31 will be a characteristic declaration for the first char, 32 will be the characteristic value, 33 will be its client configuration. 34 would be the characteristic declaration for the 2nd characteristic, 35 will be the char value, and so on.

I have found that any of the same device will have the same handle correspond to the same characteristic. eg in the above example 32 will always be the handle for the characteristic 1 value.
I have found that this is also persistent across different platforms, from Windows (using TI's BTOOL application), linux (using a generic dongle and some python scripts) to using the ESP32 as a central connecting to both ESP32's and other peripherals.

rojer, 2017-09-26:
@lucazader thanks Lucas. so, assuming i know the service and charactristic UUIDs, what is the right sequence of calls to be able to `esp_ble_gattc_write_char`? gatt_client example enumerates all the services and characteristics, so that yields handles kind of naturally, but if i want to poke specific one, what should i do?

lucazader, 2017-09-26:
Well if you dont know the device or handles ahead of time, you would have to do the discovery like in the example.

If you do know the device and its handles ahead of time, then you dont have to perform this discovery, you can just start referencing handles.

Expading on the example above. the following code would send data to characteristic 1
```
#define CHAR_1_VAL_HANDLE 32
uint8_t data[5] = {0x00, 0x01, 0x02, 0x03, 0x04};

uint16_t handle = CHAR_1_VAL_HANDLE;
ESP_LOGI("Write Data", "Writing %i bytes", length);
esp_log_buffer_hex("data_over_air", data, length); //this is a recently depreciated API call
esp_ble_gattc_write_char(gattc_if,
                         conn_id,
                         handle,
                         sizeof(data),
                         data,
                         ESP_GATT_WRITE_TYPE_NO_RSP,
                         ESP_GATT_AUTH_REQ_NONE);
```
The following code would
```
#define CHAR_1_CONFIG_HANDLE 33
#define ESP_CHAR_CONFIG_NOTIFICATION_VALUE 0b01
uint16_t handle = CHAR_1_CONFIG_HANDLE;
uint16_t notify_en = ESP_CHAR_CONFIG_NOTIFICATION_VALUE;
esp_ble_gattc_write_char_descr(gattc_if,
                               conn_id,
                               handle,
                               sizeof(notify_en),
                               (uint8_t *)&notify_en,
                               ESP_GATT_WRITE_TYPE_RSP,
                               ESP_GATT_AUTH_REQ_NONE);
```

rojer, 2017-09-26:
this is the crux of my question though: do i know handles ahead of time? they _seem_ stable, you say, but are they really? i'm pretty sure UUIDs are, and it was possible to use them directly before.
on the GATTS side, handles seem to be assigned sequentially by the stack, starting with 40 if i remember correctly, but that seems arbitrary. it may be stable in one firmware release, but nothing says it will stay like that afterwards. i am not at all convinced hard-coding handles is a good idea.
i have no problem using handles on the GATTS side, because i just record them at service registration time and i'm sure they will stay the same until service is unregistered.
on the client side though, i don't control the handles, they seem more ephemeral and if my understanding is correct, require enumeration to resolve at least once per connection.
this is what i think stack should take care of, and it was, prior to this change.

nkolban, 2017-09-26:
My testing seems to show that that the new GATTC ESP-IDF client support queries the services and characteristics and descriptors and maintains a local, in memory cache of what maps to what.  These are then queryable using:

* esp_ble_gattc_get_all_char
* esp_ble_gattc_get_all_descr
* esp_ble_gattc_get_attr_count
* esp_ble_gattc_get_db
* esp_ble_gattc_get_service

and possibly others.  So far so good (me thinks).

rojer, 2017-09-26:
@nkolban would be great if it did, however, `esp_ble_gattc_get_char_by_uuid` has the following explicit note:
```
Note: It just get characteristic from local cache, won't get from remote devices.
```
as do all the other methods.
taken at face value, it suggests that it's not enough to simply invoke it with the desired uuid to obtain a handle that can be then passed to write_char. i assume it's a cache of search results, so search needs to be performed first. further, eviction policy of this cache is unknown, so i don't know how long the results will remain valid.

nkolban, 2017-09-26:
In my unit testing (and this is not scientific), I have found that after performing an open() against the BLE Server, I was able to use the "cache".  My guess is that the cache is built internally inside the ESP-IDF at open time.  The note here is a reminder that we are getting information from the in memory cache that may have changed on the BLE Server.  Should we need to refresh the ESP-IDF RAM cache, we have another API called "esp_ble_gattc_cache_refresh" that can be explicitly called to refresh as needed.

Again ... pure guesswork on my part ... but Ive got to hold a mental model (even if wrong) to make progress and if my model is wrong, I'll be delighted to question and improve it.  So please ... question my model so that we can all learn.

rojer, 2017-09-26:
@nkolban ok, that's good to know. in that case, it's not so bad. though still, i think API made a step back here. i'll give it a go.

kp372, 2017-09-29:
`E ( ) BT: btc_gattc_call_handler()`

I am getting above runtime error while implementing [SensorTag's Client](https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLETests) app on ESP32. This error is thrown whenever the client tries to read service ID.

I have also seen this error on ESP-IDF's sample applications, to begin with. That makes me pretty certain that this is the issue on GATTC handling routines.

lucazader, 2017-10-01:
I have a gattc application working well with the new api, using the handles.

In my case I know the devices I am connecting to, so I know the handles that I need to use, so i am not doing a service discovery to make to connection process much faster and snappier.

heyinling, 2017-10-09:
@rojer the desired usage of the original API is not manually fill in the uuid structure. 

```c
typedef struct {
    esp_bt_uuid_t   uuid;                   /*!< UUID */
    uint8_t         inst_id;                /*!< Instance id */
} __attribute__((packed)) esp_gatt_id_t;
```
in `esp_gatt_id_t` , instance ID is required, in case there're some attributes with the same UUID. instance ID is allocated by the BLE stack. 

you need to call `esp_ble_gattc_get_characteristic` to get `esp_gatt_id_t`. this API is difficult to use. you need to first set `start_char_id` to NULL, get the first char id from callback event, then set the got id as `start_char_id`, it will return the next char in callback event. need to continue calling this API until you get the required char.

the new API change this async get handle procedure to sync APIs, which is easier to use.

In your scenario, the instance id is fixed value. the new API do adds some more code (several lines). but considering complicated scenarios, which need to be covered as spec allows this, the new API make things a lot easier.

rojer, 2017-10-09:
@heyinling main question is: do i need to perform explicit service/characteristic discovery to be able to resolve UUIDs to handles or will SDK do it for me?

heyinling, 2017-10-10:
calling `esp_ble_gattc_search_service` is required. it will first try to get db from local cache, if failed then it will perform service and char discovery.

Yulong-espressif, 2017-10-16:
@rojer the new gattc API is more more powerful then the old one. Perhaps at the beginning you will not adapt, but when you are familiar with it, you will find a lot more powerful than the original API, such as:
1. When you already know the server-side attribute database handle series, you do not need to call to find the service process, you can directly call the corresponding handle to read and write operations
2. The new API allows the user to get the entire server database cahe, with this cache, the user living on our API and then encapsulate the new API will become very simple.
Of course, there are many powerful features, I will not enumerate, in short, the new API will certainly be more useful than the old API and powerful, and slowly adapt to it, you will find a new continent.

rojer, 2017-10-16:
ok, i think it's clear now, i will go ahead and adapt my code to the new API. thank you1

closed 2017-10-16 by rojer

rojer, 2017-10-18:
@Yulong-espressif @heyinling one last question regarding cache persistence: if i perform a search once, can i count on entries staying in the cache while conn_id is open? what is the cache eviction policy?