espressif/esp-idf #1042: BLE: How to set icon/appearance characteristic 0x2a01

nkolban, 2017-09-26:
When we use the ESP-IDF to create a BLE Server, we find that a service with UUID 0x1800 is created for us.  We find that it has a number of pre-populated characteristics include 0x2a01 which is "Appearance".  Unfortunately, I am not able to find any ESP-IDF exposed APIs that allow us to set/change this property.

Is this an omission in the ESP-IDF BLE API set?  Am I missing some property, parameter or recipe for setting such?  (For example ... to set the Device Name characteristic (0x2a00) I would use the exposed API called `esp_ble_gap_set_device_name` ... but nothing similar is obvious for appearance/icon).

Yulong-espressif, 2017-10-16:
@nkolban Thanks for the reminder, we will add the API to set icon in the next version(v3.1) of the IDF SDK.

FayeY, 2018-04-24:
It has been merged, please have a try with the latest master. Thanks.

closed 2018-04-24 by FayeY