espressif/esp-idf #1044: libsodium: mbedtls SHA512 breaks crypto_sign

zielmicha, 2017-09-27:
In default configuration libsodium ESP-IDF component returns invalid data for `crypto_sign` module (e.g `crypto_sign_detached`. However, after removing `*sha*.{h,c}` files from ESP32 port and adding `crypto_hash/sha{256,512}/cp` back to Makefile, `crypto_sign` starts to work.

Apparently, custom libsodium SHA512 API is not implemented correctly (SHA512 works for "normal" uses).

zielmicha, 2017-09-27:
For completness: [this patch](https://gist.github.com/zielmicha/5eb22a1b69419b9eee7dd921784a645e) fixes the problem, but has undesirable effect of disabling SHA512 and SHA256 hardware acceleration.

projectgus, 2017-09-27:
Hi @zielmicha ,

Thanks very much for reporting this and sorry for the hassle. I'm able to reproduce it here and have a fix (including a new config option to disable the mbedTLS wrapping for easier testing).

Fix should be on github soon, until then the workaround you have seems like a good one.

Angus

closed 2017-10-14 by igrr

projectgus, 2017-10-16:
Thanks for your patience with this issue. 

A small additional note now this fix has landed, during testing I realised that the libsodium SHA API and ESP32 hardware accelerated SHA are incompatible.

The reason is that the mbedTLS SHA API has an mbedtls_shaxxx_free() function which needs to be called when the user is done with a SHA context. For hardware accelerated SHA sessions, we use this to free any in-progress SHA session which is loaded into the hardware unit but hasn't calculated the final digest value yet.

libsodium has no such function, the assumption is that once the SHA state structure is no longer being used the SHA session is done. As a result, if a SHA session is never finished (ie the in-progress digest is discarded without the final value being calculated), there's no way for IDF to know that the context can be removed from hardware and the hardware reused. Essentially once this happens the hardware instance is "leaked' and all remaining SHA sessions run in software.

There's no fix to this that doesn't involve either extending the libsodium API or reworking the ESP32 SHA acceleration hardware. So I've made the two features mutually exclusive in the IDF configuration.