espressif/esp-idf #1046: esp_wifi_restore() bug

samsam4, 2017-09-28:
Event handler fails to get any other events than 2: SYSTEM_EVENT_STA_START after esp_wifi_restore() . (Same is valid for system_restore etc. that are esp_wifi_restore() wrappers.)

The simplest example to illustrate the bug is with esp-idf-template-master  (just add/remove  esp_wifi_restore()  and few debub LOGI):

```
static const char *TAG = "test";`

esp_err_t event_handler(void *ctx, system_event_t *event)
{
   ESP_LOGI(TAG, ">>>event_handler: %d", event->event_id);  //ss
    return ESP_OK;
}

void app_main(void)
{
    nvs_flash_init();
    tcpip_adapter_init();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
//    ESP_LOGI(TAG, ">>>restore: ");  //ss
//    ESP_ERROR_CHECK( esp_wifi_restore());
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    wifi_config_t sta_config = {
        .sta = {
            .ssid = "XXXSSID",
            .password = "xxxpw",
            .bssid_set = false
        }
    };
    ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &sta_config) );
    ESP_ERROR_CHECK( esp_wifi_start() );
    ESP_ERROR_CHECK( esp_wifi_connect() );

```

Now the output without esp_wifi_restore() is:

> I (1311) wifi: Init rx ampdu len mblock:7
I (1315) wifi: Init lldesc rx ampdu entry mblock:4
I (1320) wifi: wifi power manager task: 0x3ffc1360 prio: 21 stack: 2560
I (1327) wifi: wifi timer task: 3ffc23dc, prio:22, stack:3584
I (1351) wifi: Init Ampdu: 1 tx baw=6 rx baw=6
I (1351) wifi: mode : sta (24:0a:c4:06:f4:98)
I (1352) test: >>>event_handler: 2
I (2076) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (2077) wifi: state: init -> auth (b0)
I (2081) wifi: state: auth -> assoc (0)
I (2090) wifi: state: assoc -> run (10)
I (2091) wifi: connected with XXXSSID, channel 6
I (2094) test: >>>event_handler: 4
I (2786) test: >>>event_handler: 7
I (12091) wifi: pm start, type:0

while output with esp_wifi_restore() is:

>I (1313) wifi: Init rx ampdu len mblock:7
I (1316) wifi: Init lldesc rx ampdu entry mblock:4
I (1321) wifi: wifi power manager task: 0x3ffc1360 prio: 21 stack: 2560
I (1327) test: >>>restore:
I (1342) wifi: wifi timer task: 3ffc15d8, prio:22, stack:3584
I (1364) wifi: Init Ampdu: 1 tx baw=6 rx baw=6
I (1364) wifi: mode : sta (24:0a:c4:06:f4:98)
I (1365) test: >>>event_handler: 2
I (2089) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (2090) wifi: state: init -> auth (b0)
I (3090) wifi: state: auth -> init (2)
I (3091) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1

liuzfesp, 2018-01-15:
@samsam4 the root cause has been found, will be fixed in v3.0

liuzfesp, 2018-01-15:
@samsam4 call esp_wifi_init() again after esp_wifi_restore() can also fixed this issue.

liuzfesp, 2018-01-15:
@samsam4 please ignore "call esp_wifi_init() again after esp_wifi_restore() can also fixed this issue." because it can't solve this issue, just wait for v3.0

liuzfesp, 2018-02-02:
HI @samsam4 is fixed in latest release/v3.0 and master branch

closed 2018-02-03 by FayeY