espressif/esp-idf #1047: GATT server denies reconnect from Android without GAP advertising

xarrange, 2017-09-28:
Hi, here is what I'm trying to do from Android:
Step 1: scan for BLE devices
Step 2: when found ESP32, read all services and their characterestics
Step 3: write one of characteristics that is defined as writeable

First and second step is completed successfully.
When the last characteristic was read I do:
```
gatt.disconnect();
gatt.close();
```
Serial output of GATT server example prints "ESP_GATTS_DISCONNECT_EVT"

After that, upon button pressing I do:
device.connectGatt(mContext, false, mWriteGattCallback);

And fait for callback to say "connected"
```
mWriteGattCallback = new BluetoothGattCallback() {
@Override
public void onConnectionStateChange(BluetoothGatt gatt, int status, int newState) {
    if (newState == BluetoothProfile.STATE_CONNECTED) {
        gatt.discoverServices();
    }
}
```
But this never occures. After approx. 115 seconds it returns disonnected with status 0x85. 
So event though server was disconnected. It doesn't allow to reconnect again.

What helped me to reconnect is to return line
`esp_ble_gap_start_advertising(&adv_params);`
to server event ESP_GATTS_DISCONNECT_EVT handler. Which I removed earlier.

I guess somehow start of advertising resets server connections.
Is it bug or feature?

lucazader, 2017-09-28:
I have not read through the BLE standard core spec doccument, however:

I think a peripheral has to be advertising for it to be connectable.
It sounds like a reasonable requirement implementation and security wise.
It also sounds like what is happening in this case.

xarrange, 2017-09-29:
Really? Probably youre'right. I was thinking that advertising is for broadcasting address. Since I obtained device address once, then I may use it several times to reconnect.

nkolban, 2017-10-05:
It is not my understanding that a BLE Server needs to be advertising in order to be able to connect to it.  However, what might be happening is that the device that is hosting the BLE Server may be using "random" Bluetooth Device Addresses.  This means that after a connection, it changes its address.  This would then be a "choice" or "function" of the BLE Server and possibly not the default but rather a specific of that instance of the BLE Server.  One might have to pair/bond in order to get a permanent address to use without having to scan again.  Have a look and see if the device address changes between scan/connect/disconnect/scan.

xarrange, 2017-10-08:
Well.. I'm talking about ESP32, not general BLE server. ESP32 doesn't change its address between reconnects. I haven't even found any function that may change it.
Moreover I don't see any point for changing address quietly in background. What if it randomly picks some other's device address? BT address has to be fixed like MAC address of NW card but could be overwritten by user

negativekelvin, 2017-10-09:
> It is not possible to connect to a peripheral which is not advertising, even though one knows its address from before. This is because the peripheral will only turn on the receiver for a set amount of time after transmitting an advertisement. This time is used to listen for connection requests and scan requests. 

https://devzone.nordicsemi.com/tutorials/21/

xarrange, 2017-10-10:
wow.. thanks. Now it's clear

Yulong-espressif, 2017-10-12:
@xarrange Do you solve your issue now?

closed 2017-10-12 by xarrange