espressif/esp-idf #1048: ULP coprocessor using C macros I2C implementation

ginodecock, 2017-09-28:
Hi,

I notice that there is no C macro to control the RTC I2C controller.

In the file ulp.h: 
_#define OPCODE_I2C 3            /*!< Instruction: read/write I2C (not implemented yet) */_

When will this be implemented? I require to control the I2C bus from the ULP processor.
It would be nice to see an example to control the RTC I2C controller.

Alternative would be to use the I2C bus via the wake up stub. Is this possible? 

Thanks

Pines, 2017-10-03:
I'm gonna need this function too,,

I have a project need power saving (which using battery, and didn't design let client charge very often), It will put to deep-sleep when no BLE or other active operation. 

but under deep-sleep, i have a I2C sensor need ULP to poll it value at a very low frequency like 10-100 Hz, and then did a little calculate or math to determine should a wake-up. 

so, at here I would like been informed any progress about implement I2C under deep-sleep.  or any suggestion help to increase my chance experiment on this "ULP I2C" function would be GREAT APPRECIATED!!!

TemcoHeng, 2017-10-23:
I2C protocol could be implemented in ULP assembly code.
This example reads the 16 bit raw temperature data from a BMP-180 sensor.
https://github.com/tomtor/ulp-i2c
FYI.

closed 2017-11-21 by FayeY