espressif/esp-idf #1051: Connecting issue with slave based on 4.1 core

kukushonok1987, 2017-09-29:
Hello
We are developing ble<-> wifi gateway and today we have faced with the following problem:
esp32 well works (in central role) with slaves which based on 4.0 and 4.2 cores. I have tested the following devices:
1. BlueNRG (core 4.0) - works well
2. BlueNRG-MS (core 4.1) doesn't work
3. EFR32BG1 (core 4.2) works well
4. Renesas chip RL78 (4.0) works well
5. Intel® Curie™ Module (4.2) works well

Please find attached sniffer logs for all mentioned devices. The common behavior for the 4.1 devices the following: after LL_Version_Ind response the exchange is stopped and event ESP_GATTC_OPEN_EVT don't occur. And after ~40 second I get disconnection with 0x0022 reason. Please help me to find the solution and please me know if I can help to figure out the cause of the problem

[ble_sniffer_logs.zip](https://github.com/espressif/esp-idf/files/1344268/ble_sniffer_logs.zip)
[image]
[image]

Thanks in

rrusak, 2017-09-29:
I have the same issue with Bluetooth based on 4.1 core, but I just switched to the devices with 4.2. It would be great to know the root of this issue. If you have solve this issue please contact me.
Thanks.

kukushonok1987, 2017-10-02:
Hello, In additional
I have commented **btsnd_hcic_rmt_ver_req** and directly call **l2cble_notify_le_connection** and now all working fine but better to figure out why ll_version response of 4.1 devices causes the stack of hang

Weijian-Espressif, 2017-10-25:
@kukushonok1987 , we will buy one BlueNRG-MS (core 4.1) and test.

kukushonok1987, 2017-10-25:
@Weijian-Espressif, Thanks a lot, please, let me know if I can help you

Weijian-Espressif, 2017-10-25:
@kukushonok1987  OK, thanks.

Weijian-Espressif, 2017-11-01:
@kukushonok1987 , I have tested  BlueNRG-MS, I build stm32 SensorDemo and flash. it works Work very well.  can you update to the latest master and retry? 

I (3757) GATTC_DEMO: searched device BlueNRG                                    
                                                                                
I (3767) GATTC_DEMO: 4f 1c c7 7f dd 00                                          
I (3767) GATTC_DEMO: searched Adv Data Len 30, Scan Response Len 0              
I (3777) GATTC_DEMO: searched Device Name Len 0                                 
I (3777) GATTC_DEMO:                                                            
                                                                                
I (3777) GATTC_DEMO: 4f 1c c7 7f dd 00                                          
I (3787) GATTC_DEMO: searched Adv Data Len 30, Scan Response Len 0              
I (3797) GATTC_DEMO: searched Device Name Len 0                                 
I (3797) GATTC_DEMO:                                                            
                                                                                
I (3797) GATTC_DEMO: stop scan successfully                                     
I (3857) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3, status 0            
I (3867) GATTC_DEMO: REMOTE BDA:                                                
I (3867) GATTC_DEMO: 02 80 e1 00 34 12                                          
I (3867) GATTC_DEMO: open success                                               
I (4607) GATTC_DEMO: ESP_GATTC_SEARCH_RES_EVT                                   
I (4607) GATTC_DEMO: 1801                                                       
I (4607) GATTC_DEMO: ESP_GATTC_SEARCH_RES_EVT                                   
I (4607) GATTC_DEMO: 1800                                                       
I (4617) GATTC_DEMO: ESP_GATTC_SEARCH_RES_EVT                                   
I (4617) GATTC_DEMO: 1b c5 d5 a5 02 00 b4 9a e1 11 3a cf 80 6e 36 02            
I (4627) GATTC_DEMO: ESP_GATTC_SEARCH_RES_EVT                                   
I (4637) GATTC_DEMO: 1b c5 d5 a5 02 00 d0 82 e2 11 77 e4 40 1a 82 42            
I (4637) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT

Weijian-Espressif, 2017-11-02:
@kukushonok1987 Can you provide your hex files?

kukushonok1987, 2017-11-02:
@ Weijian-Espressif Sorry for inconvenience, now I am in vacation I will be able to support you next week. Thanks a lot and sorry again

Weijian-Espressif, 2017-11-06:
@kukushonok1987 thanks,. Now i have a X-NUCLEO-IDB05A1 and a STM32L053 board. What should I do to recurrence this issue?

kukushonok1987, 2017-11-08:
@Weijian-Espressif  Hello, Today I have tested my app with new esp-idf and issue still reproduced:
ST provides some examples for bluenrg-ms board, could you start "SensorDemo_New" (for instance) and try to connect esp32 to bluerng-ms?

Weijian-Espressif, 2017-11-08:
@kukushonok1987  Could you provide the  download link ?

kukushonok1987, 2017-11-08:
@Weijian-Espressif please find link below, just pay attention that registration required 
https://my.st.com/content/ccc/resource/technical/software/application_sw/3a/37/97/ad/f6/47/47/d2/STSW-BLUENRG-DK.zip/files/STSW-BLUENRG-DK.zip/jcr:content/translations/en.STSW-BLUENRG-DK.zip

Weijian-Espressif, 2017-11-09:
@kukushonok1987  Could you help me to test the issue? add some logs in btm_read_remote_version_complete() in btm_acl.c, line 888,  check if this function is called and then called l2cble_notify_le_connection().  Thanks.

kukushonok1987, 2017-11-10:
@Weijian-Espressif  Sorry for delaying, now I am in busyness trip, I will provide log next week to you. Thanks a lot

kukushonok1987, 2017-11-16:
@Weijian-Espressif  I have checked function btm_read_remote_version_complete is not called, please let me know if you need any help

Weijian-Espressif, 2017-11-17:
@kukushonok1987 Thanks for your help. as you know, ESP32 as master will send btsnd_hcic_rmt_ver_req() when received feature response.Then ESP32 will send a connection success notification to the upper layer when get btm_read_remote_version_complete event.  

BlueNRG-MS send wrong ll_version_ind when exchange ble version, PDU_Length should be 6, controller throw away the wrong packet, so connection is timeout.
[image]

kukushonok1987, 2017-11-17:
@Weijian-Espressif  ooo, thanks a lot for your help!

closed 2017-11-21 by FayeY

AlpinProgrammer, 2020-05-19:
Hello,
As I understand it is an issue on the BlueNRG-MS's part.
How is it possible to solve the problem by modifying the ESP32 code ?
By commenting the following, it seems that it should resolve the issue. However, it does not solve the problem.
https://github.com/espressif/esp-idf/blob/463a9d8b7f9af8205222b80707f9bdbba7c530e1/components/bt/host/bluedroid/stack/btm/btm_ble_gap.c#L4110-L4126

Thanks in advance