espressif/esp-idf #1054: WPA2 enterprise example not working with eduroam

maur3ra, 2017-09-30:
I got myself [this ESP32 development board](https://www.amazon.de/dp/B071P98VTG/ref=pe_3044161_189395811_TE_dp_1)
and installed the toolchain as described [here](https://esp-idf.readthedocs.io/en/v2.1/get-started/macos-setup.html)

When I now use the wpa2_enterprise example, configured (make menuconfig) with WiFI SSID "eduroam", EAP method 1, my email as EAP1 ID, EAP user name (xxxxx@uni-tuebingen.de) and password as described at [my home institution](http://www.uni-tuebingen.de/en/facilities/zentrum-fuer-datenverarbeitung/dienstleistungen/netzzugang/wlan.html), this compiles and uploads: 

> WARNING: Toolchain version is not supported: 1.22.0-61-gab8375a
> Expected to see version: 1.22.0-73-ge28a011
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> WARNING: Toolchain version is not supported: 1.22.0-61-gab8375a
> Expected to see version: 1.22.0-73-ge28a011
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> Flashing binaries to serial port /dev/tty.SLAB_USBtoUART (app at offset 0x10000)...
> esptool.py v2.1
> Traceback (most recent call last):
>   File "/Users/maurera/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2524, in <module>
>     _main()
>   File "/Users/maurera/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2517, in _main
>     main()
>   File "/Users/maurera/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2246, in main
>     esp = chip_class(args.port, initial_baud)
>   File "/Users/maurera/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 177, in __init__
>     self._port = serial.serial_for_url(port)
>   File "/Library/Python/2.7/site-packages/serial/__init__.py", line 78, in serial_for_url
>     instance.open()
>   File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 289, in open
>     self.fd = os.open(self.portstr, os.O_RDWR|os.O_NOCTTY|os.O_NONBLOCK)
> OSError: [Errno 16] Resource busy: '/dev/tty.SLAB_USBtoUART'
> make: *** [flash] Error 1
> mbple-3:wpa2_enterprise maurera$ make flash
> WARNING: Toolchain version is not supported: 1.22.0-61-gab8375a
> Expected to see version: 1.22.0-73-ge28a011
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> WARNING: Toolchain version is not supported: 1.22.0-61-gab8375a
> Expected to see version: 1.22.0-73-ge28a011
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> Flashing binaries to serial port /dev/tty.SLAB_USBtoUART (app at offset 0x10000)...
> esptool.py v2.1
> Connecting......
> Chip is ESP32D0WDQ6 (revision 1)
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Auto-detected Flash size: 4MB
> Flash params set to 0x0220
> Compressed 17776 bytes to 10472...
> Wrote 17776 bytes (10472 compressed) at 0x00001000 in 0.9 seconds (effective 153.1 kbit/s)...
> Hash of data verified.
> Compressed 509920 bytes to 325366...
> Wrote 509920 bytes (325366 compressed) at 0x00010000 in 28.8 seconds (effective 141.7 kbit/s)...
> Hash of data verified.
> Compressed 3072 bytes to 82...
> Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 2051.7 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting...

but doesn't connect to eduroam: 

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0010,len:4
> load:0x3fff0014,len:5492
> load:0x40078000,len:0
> load:0x40078000,len:12180
> entry 0x40078f54
> [0;33mW (33) rtc_clk: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 40 MHz.[0m
[... 147 lines cut ...]
> [0;32mI (32317) example: GW:0.0.0.0[0m
> [0;32mI (32317) example: ~~~~~~~~~~~[0m
> I (34237) wpa: >>>>>wpa2 FIALED
> 
> I (34247) wpa: wpa2 task delete
> 
> [0;32mI (34317) example: ~~~~~~~~~~~[0m
> [0;32mI (34317) example: IP:0.0.0.0[0m
> [0;32mI (34317) example: MASK:0.0.0.0[0m
> [0;32mI (34317) example: GW:0.0.0.0[0m
> [0;32mI (34317) example: ~~~~~~~~~~~[0m
> I (35167) wifi: pm start, type:0
> 
> [0;32mI (36317) example: ~~~~~~~~~~~[0m
> [0;32mI (36317) example: IP:0.0.0.0[0m
> [0;32mI (36317) example: MASK:0.0.0.0[0m
> [0;32mI (36317) example: GW:0.0.0.0[0m
> [0;32mI (36317) example: ~~~~~~~~~~~[0m
> [0;32mI (38317) example: ~~~~~~~~~~~[0m
> [0;32mI (38317) example: IP:0.0.0.0[0m
> [0;32mI (38317) example: MASK:0.0.0.0[0m
> [0;32mI (38317) example: GW:0.0.0.0[0m
> [0;32mI (38317) example: ~~~~~~~~~~~[0m

Adding the Telekom root certificate as wpa2_ca.pem doesn't seem to change anything. Is there anything I could do, and to get more details where/how it failed?

XinDeng11, 2017-11-14:
Hi @maur3ra Sorry for later reply, have you tried with our latest IDF? What's more, have you udpate the router server's certificate in the example's main folder? There is another customer also met the problem, we suggest them update their router server's certificate and problem solved. There are two certificates wpa_server.key and wpa_server.pem, you need copy them to the router server and replace the original, after change them, rename the name to server.key and server.pem

maur3ra, 2017-11-18:
Hi @XinDeng11 , thanks for the helpful reply! You suggest changing the certificate on the router? Or Where exactly would the router server sit within the eduroam infrastructure (world-wide between schools and universities)? As far as I know, the authentication is done by the server at the "home organization" (my university), nevertheless at which partner site and in which building you are.  And I probably don't have access to the router, right? 

On eduroam, with the iPhone I just connect using username and password, on Android devices you also have to add the root certificate. Do you suggest the opposite (changing it on the server rather than getting the server certificate), or did I misunderstand you?

sagb2015, 2019-06-19:
@maur3ra - This is probably too late to ask for, but if you still face the issue with latest release, we will actively look into it. Thank you and apologies for the delay!

sagb2015, 2019-08-20:
Closing this issue because of lack of updates. Feel free to re-open if the issue is observed again.

closed 2019-08-20 by sagb2015