espressif/esp-idf #1056: Parallel Interface?

hopkinskong, 2017-09-30:
Is there anyway to read data from a parallel interface? I want to read data from an AL440B FIFO Memory.

projectgus, 2017-10-04:
There is hardware support for this kind of interface, the I2S peripheral has a "parallel" mode where instead of standard I2S it can add up to 16 parallel I/O lines.

Currently ESP-IDF doesn't have full software support for this. However there is an example to use a parallel camera interface on @igrr's personal github account, here: https://github.com/igrr/esp32-cam-demo

I'm going to leave this open as a feature request for parallel I2S support in the IDF I2S driver.

limaconaudio, 2018-02-23:
Hi! Does this mean we may see 8ch i2s parallel audio?

LRCK
BCK
DAT0  = ch1,2
DAT1  = ch3,4
DAT2  = ch5,6
DAT3  = ch7,8

jack0c, 2018-03-21:
@FayeY Seems resolved by houwenxiang.

closed 2018-10-30 by Alvin1Zhang