espressif/esp-idf #1058: ESP32 WiFi can't connect to network with hidden SSID

yaoyogi, 2017-10-01:
I am using ESP-IDF v2.1 and Espressif ESP32 Development Board - Developer Edition 

I am unable to WiFi connect to an AP with hidden SSID. When I change the router to advertise the SSID, then the code works.

liuzfesp, 2017-11-26:
@yaoyogi this is a know issue and I think it's already fixed in latest IDF.

closed 2017-11-27 by FayeY

pfe-aicoe, 2018-03-12:
I am on version 3.0 .  Still getting the WL_NO_SSID_AVAIL for the hidden ssid.

XinDeng11, 2018-03-13:
@pfe-aicoe Hi， I can't reporduce it, maybe you can tell us which channel has you set and what's your country region, beacuse we send the probe request frame according to the channel number and country region, some channels is forbidden to set probe request activily in some countries. If you set to the specfic channel fortunately, it shouldn't connect.

theproxy, 2018-03-13:
I had this issue today. It didn't work with WifiMulti as the client, but it did with the straight WiFi.h library. We didn't dig further just yet, but I wanted to add another data point.

pfe-aicoe, 2018-03-13:
Alan, Can you please share your code. I am using https://github.com/JeroenBeemster/ESP32-WPA2-enterprise/blob/master/ESP32_WPA2enterprise.ino 
This has 
#include "esp_wpa2.h"
#include "WiFi.h"

XinDeng11, 2018-03-23:
Hi @pfe-aicoe , I saw you use the wpa enterprise, so I do a test with a hidden enterprise AP, found the connect is OK. So I still suggest you checking about your channel whether is forbidden to set probe request activily. I.e, in China, we can do active scan in channel 1~11, but can not do active scan in channel 12 and 13, if a hidden SSID AP in channel 12, 13. We can't send probe request so that we wouldn't receive probe response so that we can't find it.

pfe-aicoe, 2018-03-23:
Can you please share your test code ? I will check on the channel as well.

Thank you

XinDeng11, 2018-03-25:
@pfe-aicoe Just our example