espressif/esp-idf #1060: error compiling on mac-os

skateone, 2017-10-01:
hi there. has anyone else experienced this compilation issue?

 make all
lxdialog/check-lxdialog.sh -check cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD -lncurses 
make[1]: *** No rule to make target `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/types.h', needed by `lxdialog/checklist.o'.  Stop.
make: *** No rule to make target `/app/esp/workspace/tt/build/include/config/auto.conf', needed by `/app/esp/workspace/tt/build/bootloader/bootloader.bin'.  Stop.

skateone, 2017-10-01:
i have solved this issue. i had to uninstall the latest xcode and revert to xcode V8.33 

there seems to be a problem there. anyway hope this helps anyone with similar issues

closed 2017-10-01 by skateone

igrr, 2017-10-01:
This may happen after you update OS X development tools. To fix, go to
$IDF_PATH/tools/kconfg and run "make clean". Then build your project again.