espressif/esp-idf #1061: [REQUEST]Add 32b meta data space to SPIFFS

luc-github, 2017-10-01:
Hi,
I am using arduino-esp32 and SPIFFS core is built by IDF
As ESP32 is still in development I would like to request SPIFFS is compilated using 
`#define SPIFFS_OBJ_META_LEN             32` to be able to use the SPIFFS meta data to store file creation/modification date/time as currently this feature (SPIFFS_OBJ_META_LEN) is disabled because space is set to 0.

The impact of such change is on tools and backward compatibility - so better to ask earlier than latter as it was suggested by @me-no-dev 

This value could be configurable by `make menuconfig` also but this means to add new parameter to tools like mkspiffs

The hardcode solution advantage would be that tools need update once and stay consistent after
The configurable solution may lead to incompatibilities with tools not using same data size but I guess can be handled

Also I read a new mkspiffs would be done soon, so It would be a good chance to get this change incorporated as well -  so it is good time for such request that will give more flexibility in SPIFFS

me-no-dev, 2017-10-01:
I am generally for this change. Lets discuss how that will work in reality :) like if time is not configured, what would be set as last modification or creation. As @luc-github mentioned, the current mkspiffs is not working and we can use the opportunity to add it there as well.

loboris, 2017-10-03:
I have an ESP32 spiffs implementation for ~5 months with metadata used for timestamp.
_mkspiffs_ also works without problems.
You may look at [my repository](https://github.com/loboris/ESP32_spiffs_example)

me-no-dev, 2017-10-03:
Yup :) we saw your repo and begun discussion on the topic :) needs to be implemented in IDF though so to be working on all esp32 levels :) thanks for implementing it:) i hope you will see it soon here as well

luc-github, 2017-10-14:
I did modification for IDF and Arduino to add last write time to SPIFFS and unify it for SD and SPIFFS:

https://github.com/luc-github/esp-idf/commit/bfc9d60bce70145e33ad78c364008b1c0ee5be01

https://github.com/luc-github/arduino-esp32/commit/4b32c261717c1db526d3548776574ffa8bb583b7 and https://github.com/luc-github/arduino-esp32/commit/2d6d6ba1e9eb61f4a99246d998e2ab4d4e86fdc2

For mkspiffs it just need to change the meta len for ESP32 package
 `#define SPIFFS_OBJ_META_LEN (4)`

So compilation is not an issue, but I wonder how to discriminate mkspiffs for ESP8266 vs ESP32 ?
Currently there is nothing but version to say which version is for ESP32

luc-github, 2017-10-16:
I did same modification for SPIFFS for ESP8266 : https://github.com/luc-github/Arduino/commit/c6bf371a412c35302945bb0ca9a9250ab6e7bd73
if it can help to simplify global consistency

luc-github, 2017-10-17:
I did PR as suggested by @me-no-dev 
idf : https://github.com/espressif/esp-idf/pull/1128 (signed twice  CLA but seems not recognize me :sob: )
esp-32 : https://github.com/espressif/arduino-esp32/pull/738
esp8266: https://github.com/esp8266/Arduino/pull/3730
to lower impact on mkspiffs
Thank you

igrr, 2018-01-08:
Merged in c03b068931b484990bbee55299502eea4c02994a and 58f046715ef925b77af0180d1a61db1d6d12cac3, thanks!

closed 2018-01-08 by igrr

JimPos, 2018-01-11:
So when might we see an example of SPIFFS with time stamp?

luc-github, 2018-01-11:
@JimPos  it is here no ?  https://github.com/espressif/esp-idf/blob/master/components/spiffs/test/test_spiffs.c#L509-L549

JimPos, 2018-01-12:
@luc-github Thanks for the reply.
I guess I'll have to learn how to use the esp-idf.
I am a lazy hobbyist that has gotten used to other people doing it for me.
I have been using the Arduino IDE because it was so easy.
I currently have no idea how to use the esp-idf or how/when esp-idf features get transferred to Arduino.
I suppose it is time for me to put in some effort.
Thank you for all your effort getting this feature added.

luc-github, 2018-01-12:
The arduino-esp32 PR https://github.com/espressif/arduino-esp32/pull/738 is still pending as it was waiting for IDF part to be done first -  I think it should be done soon after the IDF part is integrated - sample is available also in arduino-esp32 part - just wait final integration by @me-no-dev ^_^

JimPos, 2018-01-12:
Great, Thanks. I anxiously await.