espressif/esp-idf #1062: I2S not usable for 8 bit samples

gamer-cndg, 2017-10-01:
When configuring the I2S peripheral using `i2s_set_clk`, the function will always fail when requested to use 8 bit samples, although this is a valid `i2s_bits_per_sample_t` value.

Example:
`i2s_set_clk(I2S_NUM_0, 44100, I2S_BITS_PER_SAMPLE_8BIT, I2S_CHANNEL_MONO);`

Will throw error
`I2S: Invalid bits per sample` 

Which is due to this check: https://github.com/espressif/esp-idf/blob/master/components/driver/i2s.c#L179

```
if (bits % 8 != 0 || bits > I2S_BITS_PER_SAMPLE_32BIT || bits < I2S_BITS_PER_SAMPLE_16BIT) {
   ESP_LOGE(I2S_TAG, "Invalid bits per sample");
   return ESP_ERR_INVALID_ARG;
}
```
Is this just a wrong `if` check or are 8 bit samples not usable with the ESP32's I2S peripheral?

hchaudhary1, 2018-06-13:
seeing same problem with released idf v3

gamer-cndg, 2018-06-13:
Yep, different line number, same code.. (Updated description)

gecko242, 2018-11-06:
Any more info on this issue? Still present in 3.2

nevercast, 2019-06-11:
I'm still seeing this!

wilhelmw201, 2020-07-15:
Hi,

please kindly pardon the delay in this response :-(. Technically the hardware supports 8-bit mode. The hardware takes a 32-bit value, puts the 7:0 bits on the right channel, and the 23:16 bits on the left channel. (Vice versa in receive mode) The rest are discarded.

This way of sending user data and storing the retrieved data is wasteful though. As a result, the allowed modes are limited to 32, 24(this is also kind of wasteful, 8 bits of 32 are discarded) and 16.

If you need the 8-bit mode, you may remove the error check. Everything should correctly function, except you may have to manually clean up your received data if you are using i2s_read.

Alvin1Zhang, 2020-07-22:
Thanks for reporting, and sorry for slow turnaround, feel free to reopen, thanks.

closed 2020-07-22 by Alvin1Zhang