espressif/esp-idf #1063: SPP Examples

PhilColbert, 2017-10-02:
Thanks for the SPP Examples, any documentation on how they work and how to set up sending data ?

Thanks

Spritetm, 2017-10-10:
What do you mean by SPP? Serial Port Profile, Standard Parallel Port, ...?

chanibal, 2017-11-13:
How can we run the SPP examples? All I'm getting when I run are two not very verbose logs:

Server:
```
W (66) rtc_clk: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 40 MHz.
I (36) boot: ESP-IDF v3.0-dev-1028-g2c95a77c 2nd stage bootloader
I (36) boot: compile time 14:07:52
I (44) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (47) boot: SPI Mode       : DIO
I (51) boot: SPI Flash Size : 4MB
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2dadc (187100) map
I (166) esp_image: segment 1: paddr=0x0003db04 vaddr=0x3ffc0000 size=0x0250c (  9484) load
I (170) esp_image: segment 2: paddr=0x00040018 vaddr=0x400d0018 size=0x69898 (432280) map
0x400d0018: _stext at ??:?

I (322) esp_image: segment 3: paddr=0x000a98b8 vaddr=0x3ffc250c size=0x0055c (  1372) load
I (323) esp_image: segment 4: paddr=0x000a9e1c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at D:/Soft/ESP32/esp-idf/components/freertos/xtensa_vectors.S:1685

I (329) esp_image: segment 5: paddr=0x000aa224 vaddr=0x40080400 size=0x125c0 ( 75200) load
I (369) esp_image: segment 6: paddr=0x000bc7ec vaddr=0x400c0000 size=0x00000 (     0) load
I (380) boot: Loaded app from partition at offset 0x10000
I (380) boot: Disabling RNG early entropy source...
I (381) cpu_start: Pro cpu up.
I (384) cpu_start: Starting app cpu, entry point is 0x40080f1c
0x40080f1c: call_start_cpu1 at D:/Soft/ESP32/esp-idf/components/esp32/cpu_start.c:222

I (0) cpu_start: App cpu up.
I (395) heap_init: Initializing. RAM available for dynamic allocation:
I (401) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (408) heap_init: At 3FFCA398 len 00015C68 (87 KiB): DRAM
I (414) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (420) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (426) heap_init: At 400929C0 len 0000D640 (53 KiB): IRAM
I (433) cpu_start: Pro cpu start user code
I (115) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (176) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (436) phy: phy_version: 362.0, 61e8d92, Sep  8 2017, 18:48:11, 0, 0
I (436) GATTS_SPP_DEMO: app_main init bluetooth

I (576) GATTS_SPP_DEMO: EVT 0, gatts if 3

E (576) GATTS_SPP_DEMO: event = 0

I (576) GATTS_SPP_DEMO: gatts_profile_event_handler 495

I (576) GATTS_SPP_DEMO: gatts_profile_event_handler 498

I (586) GATTS_SPP_DEMO: gatts_profile_event_handler 501

I (596) GATTS_SPP_DEMO: EVT 22, gatts if 3

E (596) GATTS_SPP_DEMO: event = 16

I (596) GATTS_SPP_DEMO: The number handle =b

E (606) GATTS_SPP_DEMO: GAP_EVT, event 4

I (606) GATTS_SPP_DEMO: EVT 12, gatts if 3

E (616) GATTS_SPP_DEMO: event = c

[... 44 lines cut ...]
I (366) boot: Disabling RNG early entropy source...
I (367) cpu_start: Pro cpu up.
I (371) cpu_start: Starting app cpu, entry point is 0x40080ec4
0x40080ec4: call_start_cpu1 at D:/Soft/ESP32/esp-idf/components/esp32/cpu_start.c:222

I (0) cpu_start: App cpu up.
I (381) heap_init: Initializing. RAM available for dynamic allocation:
I (388) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (394) heap_init: At 3FFCA2B0 len 00015D50 (87 KiB): DRAM
I (400) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (406) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (413) heap_init: At 400920F4 len 0000DF0C (55 KiB): IRAM
I (419) cpu_start: Pro cpu start user code
I (101) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (152) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (422) phy: phy_version: 362.0, 61e8d92, Sep  8 2017, 18:48:11, 0, 0
I (422) GATTC_SPP_DEMO: app_main init bluetooth
```

From this, I have no idea what to do next, when I reset any of these devices, nothing shows up on the other one, so they are probably not seeing each other.

Also some lines are red so maybe an error of some kind?

Could someone from Espressif please look and say if these examples are even working at the moment, please?

xiewenxiang, 2017-11-28:
@chanibal 
because the API was modified in IDF 3.0 , so , the client demo need refactor.
we have refactor the ble_spp_client demo, but it is not merged into master, we will MR it ASAP.

FayeY, 2018-01-11:
Hi, this issue should have been solved in the latest IDF, please have a try. If it is not, please feel free to reopen it. Thanks.

closed 2018-01-11 by FayeY