espressif/esp-idf #1064: Rebooting again and again and finally no reboot

CPZbre, 2017-10-02:
Hi !

After an Unhandled debug exception the ESP reboot many time and finally blocked on rebooting...

> ***ERROR*** A stack overflow in task tTask has been detected.
> aboGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
> Debug exception reason: BREAK instr
> Register dump:
> PC      : 0x400d3170  PS      : 0x00060736  A0      : 0x80084964  A1      : 0x3ffb4984
> 0x400d3170: esp_vApplicationIdleHook at C:/msys32/home/chris/esp/esp-idf/components/esp32/freertos_hooks.c:52
> 
> A2      : 0x00000002  A3      : 0xffffffe0  A4      : 0x00000000  A5      : 0x3ffbbcb0
> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000
> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x40086455  A13     : 0x3ffbbc90
> 0x40086455: _xt_context_save at ??:?
> 
> A14     : 0x00000026  A15     : 0x00000014  SAR     : 0x0000000e  EXCCAUSE: 0x00000001
> EXCVADDR: 0xffffffe0  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
> 
> Backtrace:rtGuru Meditation Error: Core  1 panic'ed (Double exception)
> Register dump:
> PC      : 0x400878a9  PS      : 0x00050036  A0      : 0x80084964  A1      : 0x40000048
> 0x400878a9: panicHandler at C:/msys32/home/chris/esp/esp-idf/components/esp32/panic.c:519
> 
> A2      : 0x00000020  A3      : 0x00000004  A4      : 0x00060723  A5      : 0x3ffb0b60
> A6      : 0x0000000a  A7      : 0x3ffbff90  A8      : 0x800fce25  A9      : 0x3ffbff60
> A10     : 0x3ffb494c  A11     : 0x3ffbff70  A12     : 0x00040025  A13     : 0x3ffb4804
> A14     : 0x3ffb4874  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000002
> EXCVADDR: 0x4000003c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x400878a9:0x40000048
> 0x400878a9: panicHandler at C:/msys32/home/chris/esp/esp-idf/components/esp32/panic.c:519
> 
> 
> Rebooting...
> ()Guru Meditation Error of type IllegalInstruction occurred on core  1. Exception was unhandled.
> Register dump:
> PC      : 0x40081116  PS      : 0x00060633  A0      : 0x8008789c  A1      : 0x3ffffea8
> 0x40081116: esp_restart_noos at C:/msys32/home/chris/esp/esp-idf/components/esp32/system_api.c:343
> 
> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3fffff28
> A6      : 0x3ffb22dc  A7      : 0x3ffb22e8  A8      : 0x400d29d8  A9      : 0x3ffffe88
> 0x400d29d8: esp_dport_access_int_deinit at C:/msys32/home/chris/esp/esp-idf/components/esp32/dport_access.c:180
> 
> A10     : 0x00000000  A11     : 0x8001f800  A12     : 0x00000020  A13     : 0x3ffb4784
> A14     : 0x3ffb48c4  A15     : 0x00000002  SAR     : 0x0000001f  EXCCAUSE: 0x00000000
> EXCVADDR: 0x4000003c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x40081116:0x3ffffea8
> 0x40081116: esp_restart_noos at C:/msys32/home/chris/esp/esp-idf/components/esp32/system_api.c:343
> 
> 
> Rebooting...
>  wGuru Meditation Error of type IllegalInstruction occurred on core  1. Exception was unhandled.
> Register dump:
> PC      : 0x40081116  PS      : 0x00060033  A0      : 0x8008789c  A1      : 0x3ffffd28
> 0x40081116: esp_restart_noos at C:/msys32/home/chris/esp/esp-idf/components/esp32/system_api.c:343
> 
> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3ffffda8
> A6      : 0x3ffb22dc  A7      : 0x3ffb22e8  A8      : 0x400d29d8  A9      : 0x3ffffd08
> 0x400d29d8: esp_dport_access_int_deinit at C:/msys32/home/chris/esp/esp-idf/components/esp32/dport_access.c:180
> 
> A10     : 0x00000000  A11     : 0x8001f800  A12     : 0x3ff40000  A13     : 0x0000007d
> A14     : 0x00000055  A15     : 0x3ffb4674  SAR     : 0x0000001a  EXCCAUSE: 0x00000000
> EXCVADDR: 0x4000003c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x40081116:0x3ffffd28
> 0x40081116: esp_restart_noos at C:/msys32/home/chris/esp/esp-idf/components/esp32/system_api.c:343
> 
> 
> Rebooting...

Again and again with different values

> Guru Meditation Error of type IllegalInstruction occurred on core  1. Exception was unhandled.
> Register dump:
> PC      : 0x40081116  PS      : 0x00060033  A0      : 0x8008789c  A1      : 0x3fffdf28
> 0x40081116: esp_restart_noos at C:/msys32/home/chris/esp/esp-idf/components/esp32/system_api.c:343
> 
> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3fffdfa8
> A6      : 0x3ffb22dc  A7      : 0x3ffb22e8  A8      : 0x400d29d8  A9      : 0x3fffdf08
> 0x400d29d8: esp_dport_access_int_deinit at C:/msys32/home/chris/esp/esp-idf/components/esp32/dport_access.c:180
> 
> A10     : 0x00000000  A11     : 0x8001f800  A12     : 0x3ff40000  A13     : 0x0000007d
> A14     : 0x00000002  A15     : 0x3fffe2a8  SAR     : 0x00000016  EXCCAUSE: 0x00000000
> EXCVADDR: 0x4000003c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x40081116:0x3fffdf28
> 0x40081116: esp_restart_noos at C:/msys32/home/chris/esp/esp-idf/components/esp32/system_api.c:343
> 
> 
> Rebooting...
> 
> 
> Backtrace: 0x40081116:0x3fffdda8
> 0x40081116: esp_restart_noos at C:/msys32/home/chris/esp/esp-idf/components/esp32/system_api.c:343
> 
> 
> Rebooting...
> 
> Backtrace: 0x40081116:0x3fffdc28
> 0x40081116: esp_restart_noos at C:/msys32/home/chris/esp/esp-idf/components/esp32/system_api.c:343
> 
> 
> Rebooting...
> 
> Backtrace: 0x40081116:0x3fffdaa8
> 0x40081116: esp_restart_noos at C:/msys32/home/chris/esp/esp-idf/components/esp32/system_api.c:343
> 
> 
> Rebooting...
And never restarts except by doing a physical reset

projectgus, 2017-10-04:
Hi @CPZbre ,

Do you get this no matter what code you run on the ESP32, or do you think there is a particular software crash which triggers it?

Unless there's particular code involved, usually this kind of instability can be traced to problems with the power supply or problems with the supporting hardware on the PCB or module. Is the brownout detector disabled (it's enabled by default)?

Angus

CPZbre, 2017-10-12:
This happens with different code and exceptionally (once a week).
Our worry is not the crash but the fact that it does not restart on its own.
It is a big problem if this happens on site, because only human intervention can restart it.
Is there a software or hardware solution to avoid this situation.

Genius-SexyBoy, 2017-11-29:
Hi @CPZbre 
I meet this problem the like you. Do you have any ways to solve it?

igrr, 2017-12-08:
Hi @Genius-SexyBoy, can you share the code to reproduce the issue?

CPZbre, 2017-12-12:
The problem is not triggered in the application code but in the bootloader in case of crash.
We often have this problem when we use the "rmt_write_items" function during the wifi connection
The use of these 2 features at the same time seems tricky and cause frequent crashes (this is not the subject) but must leave MCU register in wrong state during reboot.

projectgus, 2019-03-01:
Hi @CPZbre ,

Sorry noone got back to about this issue. Some bugs were fixed last year in the "soft restart" behaviour where certain peripheral state could remain incorrect during reboot, as you point out.

If you're still seeing the problem with more recent IDF and you have some code to reproduce it, please let us know and we'll reopen.

closed 2019-03-01 by projectgus