espressif/esp-idf #1065: PPPOS Protocol Reject (IPV6)

HardwireIO, 2017-10-02:
Hi!

I'm trying to test PPPOS using a GSM modem connected to a serial port.
Below ppp logs:

```
ppp phase changed[1]: phase=0
ppp_connect[1]: holdoff=0
ppp phase changed[1]: phase=1
pppos_connect: unit 1: connecting
ppp_start[1]
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=FF FF FF FF
ppp_recv_config[1]
ppp: auth protocols: PAP=1 CHAP=0 CHAP_MD5=0
pppos_write[1]: len=24
ppp_start[1]: finished
pppos_input[1]: got 72 bytes
pppos_write[1]: len=22
pppos_write[1]: len=24
pppos_input[1]: got 46 bytes
netif_set_mtu[1]: mtu=1500
pppos_send_config[1]: out_accm=0 0 A 0
ppp_send_config[1]
pppos_recv_config[1]: in_accm=0 0 A 0
ppp_recv_config[1]
ppp phase changed[1]: phase=5
pppos_write[1]: len=10
pppos_input[1]: got 26 bytes
PAP authentication succeeded
ppp phase changed[1]: phase=7
pppos_write[1]: len=32
pppos_write[1]: len=18
pppos_write[1]: len=14
pppos_input[1]: got 43 bytes
pppos_write[1]: len=26
Protocol-Reject for 0x8057 received
pppos_input[1]: got 11 bytes
LCP terminated by peer
ppp phase changed[1]: phase=4
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=0 0 A 0
ppp_recv_config[1]
pppos_write[1]: len=8
pppos_input[1]: got 13 bytes
pppos_input[1]: got 12 bytes
ppp phase changed[1]: phase=10
Connection terminated.
ppp phase changed[1]: phase=0
ppp_link_terminated[1]
ppp_end[1]

```
I got a "Protocol Reject for 0x8057" that should be related to IPV6. 
Is this the reason for a connection closed?

I know that IPV6 is enabled by default on LWiP, so the message "Protocol-Reject received" let me think problem is on the other side.

From pppos debug logs I cannot figure out the reason.

loboris, 2017-10-03:
It is not the reason for connection close. It only means that the other side does not accept IPv6 protocol.
You have _LCP terminated_ by peer message, the reason could be in your GSM modem connection/configuration.
I'm also having this reject message, but the connection is successful:

```
ppp phase changed[0]: phase=0
ppp_connect[0]: holdoff=0
ppp phase changed[0]: phase=1
pppos_connect: unit 0: connecting
ppp_start[0]
pppos_send_config[0]: out_accm=FF FF FF FF
ppp_send_config[0]
pppos_recv_config[0]: in_accm=FF FF FF FF
ppp_recv_config[0]
ppp: auth protocols: PAP=1
pppos_write[0]: len=24
ppp_start[0]: finished
pppos_input[0]: got 72 bytes
pppos_write[0]: len=22
pppos_write[0]: len=24
pppos_input[0]: got 46 bytes
netif_set_mtu[0]: mtu=1500
pppos_send_config[0]: out_accm=0 0 A 0
ppp_send_config[0]
pppos_recv_config[0]: in_accm=0 0 A 0
ppp_recv_config[0]
ppp phase changed[0]: phase=5
pppos_write[0]: len=10
pppos_input[0]: got 21 bytes
PAP authentication succeeded
ppp phase changed[0]: phase=7
pppos_write[0]: len=32
pppos_write[0]: len=18
pppos_write[0]: len=14
pppos_input[0]: got 60 bytes
pppos_write[0]: len=26
Protocol-Reject for 0x8057 received
pppos_write[0]: len=14
pppos_input[0]: got 10 bytes
pppos_write[0]: len=8
pppos_input[0]: got 28 bytes
pppos_write[0]: len=26
pppos_input[0]: got 28 bytes
Could not determine remote IP address: defaulting to 10.64.64.64
sifvjcomp[0]: VJ compress enable=0 slot=0 max slot=0
sifup[0]: err_code=0
local  IP address 10.227.55.119
remote IP address 10.64.64.64
primary   DNS address 195.29.247.161
secondary DNS address 195.29.247.162
ppp phase changed[0]: phase=8

```

imotion-software, 2017-10-03:
Ok great @loboris, thank you for your answer. 
It's not easy to figure out the reason from the those logs, so I simply figured out it was due to this error message.

I will check flow control on serial port between ESP32 and GSM, because I cannot think about any other reason if error is caused by GSM.

Do you know if there's any other software or configuration that can bring to "LCP terminated by peer", please?

loboris, 2017-10-03:
You can try to use different authentication protocol with `ppp_set_auth(ppp_pcb *pcb, u8_t authtype, const char *user, const char *passwd);`
_PPPAUTHTYPE_NONE, PPPAUTHTYPE_PAP, PPPAUTHTYPE_CHAP, PPPAUTHTYPE_MSCHAP, 
PPAUTHTYPE_MSCHAP_V2, PPPAUTHTYPE_EAP, PPPAUTHTYPE_ANY_
and check the user name and password.
Make shure that you have enabled the protocol in menuconfig...

HardwireIO, 2017-10-03:
Hi @loboris,
I found the problem. It was not related to IPV6 but due to the fact we launched PPP session once modem was not registered to the network. 

Now it looks like PPP is working and we got IP address successfully, but there's no way to redirect traffic to PPP interface. I think it's due to the fact we have wifi on (in AP mode but not connected).

How can we choose what is the right interface for the network traffic?
Is there anything equivalent to Linux "route" command?

HardwireIO, 2017-10-03:
Maybe I already got the answer:

`netif_set_default` 

Is it correct?

loboris, 2017-10-03:
I'm also interested in routing wifi to ppp, but I've had no success so far.
I've tried with `netif_set_default` but it doesn't change anything.

I'm using gsm to connect to the Internet and wifi (AP) only for local configuration, so it is not a big problem, but I'd like to find a solution.

HardwireIO, 2017-10-03:
@loboris actually I'm trying to do the same.
WiFi for in AP mode for local configuration and PPPoS for Internet connection but after PPP negotiation completes and we got an IP address, all socket already created are not able to communicate towards PPP interface at all.

Actually, all sockets are created at boot when PPP is down, and I'd like to route all those sockets to PPP interface, but netif_set_default seems not to do this job.

Do I have to close and reopen all sockets again?

loboris, 2017-10-04:
I've added the example of using WiFi in AP mode and GSM/PPPoS for Internet connection to my [ESP32-PPPOS-EXAMPLE](https://github.com/loboris/ESP32-PPPOS-EXAMPLE) repository.
Simple web server is included to test WiFi connection while some test pages are accessed from Internet via GSM/PPPoS.
For some reason PPPoS must be started first, then WiFi/AP (check **pppos_client_main.c**).
With such configuration I can successfully get the response from WebServer while accessing files from Internet via GSM.

liuzfesp, 2018-03-27:
@HardwireIO , can @loboris 's suggestion help to solve your issue?

crismansantos, 2018-04-13:
Hi!
@loboris and @HardwireIO , did you route wifi ap to pppos interface?
I'm trying to do it but isnt work
Can you help-me?

liuzfesp, 2018-06-22:
Hi @crismansantos, currently we don't provide support for PPP in short term.

closed 2020-03-06 by espressif-bot