espressif/esp-idf #1068: ESP-IDF GATT CLIENT demo

pbrito55, 2017-10-02:
Hi I'm having this error:

CORRUPT HEAP: Bad tail at 0x3fffee40. Expected 0xbaad5678 got 0xbaad2902
assertion "head != NULL" failed: file "/Users/ficeto/Espressif/ESP32/esp-idf/components/heap/./multi_heap_poisoning.c", line 195, function: multi_heap_free
abort() was called at PC 0x40147ebf on core 0

Backtrace: 0x40089db0:0x3fff14a0 0x40089eaf:0x3fff14c0 0x40147ebf:0x3fff14e0 0x40089af1:0x3fff1510 0x4008512a:0x3fff1530 0x400855a1:0x3fff1550 0x4000bec7:0x3fff1570 0x400d3708:0x3fff1590 0x400d3c19:0x3fff1610 0x400dfa45:0x3fff1640 0x400dfdf6:0x3fff1680 0x400dd18d:0x3fff16b0

Rebooting...

While running the the example  GATT CLIENT demo.(esp-idf/examples/bluetooth/gatt_client/main/gattc_demo.c)

This happen in code line 195 & 249 while doing the free(descr_elem_result); and free(char_elem_result);

If I comment this lines the example just starts working. This is strange to me because  if we have a malloc we should have then a free but he looks like he is complaining like we already had done the free.

Has I'm using the example just to interface a ble server I'm afraid if I do not do the free then I will start having memory leak.

My SW verssion  Sep 29, 2017   e6afe28
Sorry about my English... 

Thanks //PB

pbrito55, 2017-10-16:
I'm already doing that.... but i looks that something strange is there I do not like when I do not  understand... why it is doing strange things  special with  dynamic allocation.

gerbert, 2017-10-16:
I'll tell you what, I had a trouble with one simple expression that looked like this (on master branch of ESP-IDF), comparison was done in a standalone task that was reading the data from the UART interface:
if (a + b < c) {
...
}
(i even tried ((a + b) < c) with the same result)
- it crashed (system hang and reboot)! Variable types where correct 100%, as well as values. The solution, in my case was :):
uint16_t k = a + b;
if (k < c) {
...
}
Which worked :). Later I switched to a branch 2.1 which is stable enough... in compare to master branch. Even now I don't what have happened. Maybe some stack issues or else...

Yulong-espressif, 2017-10-17:
@pbrito55 Can you show me the code, I didn't have the same trouble with you in my environment. So I do not know how to be able to reproduce your problem. Thanks.

pbrito55, 2017-10-17:
I was using this example in esp-arduino and got the reboot. Then decided to try it in esp-idf.

Arduino esp has heap corruption detection active. so I activated also in esp-idf by make menuconfig  
(if the heap corruption is not active the there is no error in esp-idf but there is no detection also...)

The esp complains about corruption when we do the free of the memory.  

 Heap corruption detection (Comprehensive)  --->              │ │
  │ │    [*] Enable heap tracing                                          │ │
  │ │    (2)   Heap tracing stack depth  
 

and made 4 changes in the example  code.

#define REMOTE_SERVICE_UUID        0x1816
#define REMOTE_NOTIFY_CHAR_UUID    0x2a5b

static const char remote_device_name[] = "Bike Combo Sensor";

and in line 274 
 /* esp_ble_gattc_write_char( gattc_if,
                                  gl_profile_tab[PROFILE_A_APP_ID].conn_id,
                                  gl_profile_tab[PROFILE_A_APP_ID].char_handle,
                                  sizeof(write_char_data),
                                  write_char_data,
                                  ESP_GATT_WRITE_TYPE_RSP,
                                  ESP_GATT_AUTH_REQ_NONE); */

Thanks for your support. And if I did not explain myself correctly please ask again as my English ... 
Also the heap corruption functionality is not my speciality and I could be doing something wrong.

But the fact is that this is activated in the arduino and I do not know how to disable it if there is no problem in the code.
And form my point of view the code of the example is correct if we have a problem must be in the GATTC API. I gess...

Once more thanks for the help.
 
Here is the full code:

`// Copyright 2015-2017 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/****************************************************************************
*
* This file is for gatt client. It can scan ble device, connect one device.
* Run the gatt_server demo, the client demo will automatically connect to the gatt_server demo.
* Client demo will enable gatt_server's notify after connection. Then the two devices will exchange
* data.
*
****************************************************************************/

#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include "nvs.h"
#include "nvs_flash.h"
#include "controller.h"

#include "bt.h"
[... 415 lines cut ...]
    ret = esp_ble_gap_register_callback(esp_gap_cb);
    if (ret){
        ESP_LOGE(GATTC_TAG, "%s gap register failed, error code = %x\n", __func__, ret);
        return;
    }

    //register the callback function to the gattc module
    ret = esp_ble_gattc_register_callback(esp_gattc_cb);
    if(ret){
        ESP_LOGE(GATTC_TAG, "%s gattc register failed, error code = %x\n", __func__, ret);
        return;
    }

    ret = esp_ble_gattc_app_register(PROFILE_A_APP_ID);
    if (ret){
        ESP_LOGE(GATTC_TAG, "%s gattc app register failed, error code = %x\n", __func__, ret);
    }
    esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
    if (local_mtu_ret){
        ESP_LOGE(GATTC_TAG, "set local  MTU failed, error code = %x", local_mtu_ret);
    }

}

`

Weijian-Espressif, 2017-10-23:
@pbrito55 our client demo has a malloc error, maybe it is the reason. the line 173 "char_elem_result = (esp_gattc_char_elem_t *)malloc(sizeof(char_elem_result) * count);" changed to "char_elem_result = (esp_gattc_char_elem_t *)malloc(sizeof(esp_gattc_char_elem_t) * count);", please have a try.

pbrito55, 2017-10-24:
Changed lines :
line173:
char_elem_result = (esp_gattc_char_elem_t *)malloc(sizeof(esp_gattc_char_elem_t) * count);
line 219:
descr_elem_result = (esp_gattc_descr_elem_t *)malloc(sizeof(esp_gattc_descr_elem_t) * count);

No more CORRUPT HEAP :-) thanks :+1: