espressif/esp-idf #1070: Clash with duplicate definitions of fcntl()

dmcnaugh, 2017-10-03:
I am trying to use `fcntl()` by including either `<fcntl.h>` or `<sys/fcntl.h>` 
But during linking I receive the following error:

```
/Users/dmcnaugh/esp/proj/build/vfs/libvfs.a(vfs.o): In function `fcntl':
/Users/dmcnaugh/esp/esp-idf/components/vfs/./vfs.c:477: multiple definition of `fcntl'
/Users/dmcnaugh/esp/esp-idf/components/newlib/lib/libc.a(lib_a-sysfcntl.o):/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/syscalls/../../../.././newlib/libc/syscalls/sysfcntl.c:12: first defined here
```

Meanwhile I am using the `linenoise` library and am aware that  `linenoise.c` includes `<sys/fcntl.h>` and uses `fcntl()` and this produces no errors during compiling or linking.

Either there is a problem with the way I am including the header files, but I have tried many combinations, or there is a problem with the way the header files are structured.

[PS. I can hack functions to do what I want to with `fcntl()` into `linenoise.c` and this works!
But this is not a longterm solution.]

jcmvbkbc, 2017-10-03:
> Either there is a problem with the way I am including the header files, but I have tried many combinations, or there is a problem with the way the header files are structured.

The error that you have is a link-time error, it has nothing to do with header files. You have it because the function fcntl is defined both in the esp-idf/components/vfs/vfs.c and in the libc and somehow you need to link with both.

dmcnaugh, 2017-10-03:
Ok thanks for narrowing it down for me.
So I think it is the authors of the ESP-IDF that need to resolve the linking problem as they are responsible for `fcntl()` being defined in two places.

projectgus, 2017-10-03:
Hi @dmcnaugh ,

Thanks for reporting this. It does look like some symbols were left in the libc which shouldn't be there.

I have a fix for this here, but it probably won't be merged until after the holiday in China ends next week (and there will be a bit of a backlog!)

As a temporary workaround, you can navigate to the `$IDF_PATH/components/newlib/lib` directory and run this command:
```
for lib in libc*.a; do ar d $lib lib_a-sysfcntl.o; done
```

And you should be able to link with the VFS fcntl().

dmcnaugh, 2017-10-04:
Thanks, but the problem must run a little deeper...

When I followed your instructions (above) I received the following output:
```
$ for lib in libc*.a; do ar d $lib lib_a-sysfcntl.o; done
ar: lib_a-sysfcntl.o: not found in archive
ar: lib_a-sysfcntl.o: not found in archive
ar: lib_a-sysfcntl.o: not found in archive
```
I understand what this is meant to do, but strangely `lib_a-sysfcntl.o` is not present in these library archives.

projectgus, 2017-10-04:
That's odd, I just ran this again on my copy of the master branch and it works. Is it possible you ran it twice? Confusingly, it produces no output when it succeeds but will produce that error output if you run it a second time (as the symbol files have already been deleted).

projectgus, 2017-10-04:
Oops, maybe you need the cross-toolchain version of ar (using the host version worked on my system but that's no guarantee it will works on yours!)

```
for lib in libc*.a; do xtensa-esp32-elf-ar d $lib lib_a-sysfcntl.o; done
```

dmcnaugh, 2017-10-04:
Thank you, yes,  your solution works.
I can now link successfully with references to `fcntl()` correctly resolved.
Thank you for your help and perseverance in providing a solution.

closed 2017-10-18 by igrr