espressif/esp-idf #1071: Turning the chip off

CristiCimpianu, 2017-10-03:
Sometime ago it used to work to use gpio_set_level to set CHIP_PU low and the chip went off.
Today I retested this functionality and:

1. First the brownout handler came in and reset the chip
2. After disabling brownout the chip stays off for ~3 seconds and resets with the reset reason RTCWDT_RTC_RESET.

I tried different ways to disable the RTCWDT (replicating rtc_wdt_disable and something I found in the bootloader) before turning the chip off, but the chip still resets.
Is there some new recommended way to programatically turn the chip off?

projectgus, 2017-10-03:
Hi @CristiCimpianu ,

I'm not sure we have a supported way to have the chip disable itself, without at least some external circuitry. What do you have connected to the EN pin apart from the GPIO? Can you please post a snippet of schematic?

CristiCimpianu, 2017-10-04:
Hi @projectgus and thanks for your quick reply.
After a few more discussions with my hardware engineer it seems that it wasn't CHIP_PU that he connected to GPIO26 for this turn off mechanism. He simply connected pin 15 to a transistor and when setting GPIO26 low the transistor cuts the main VCC line from the battery.
I have no clue why this worked a couple months ago, but if you don't know a supported way to turn the chip off, then please close this issue with my apologies for the wasted time.

projectgus, 2017-10-05:
No problems. We don't have a supported way to do this. You can probably do it via a transistor that disables either VCC or EN, but you may need to add some latching circuitry to make sure the pin stays there and doesn't bounce around while the chip is powering down. Also a pulldown on GPIO26, for similar reasons.

There's also deep sleep mode, depending on how low the power requirements are while the chip is "off".

I'm going to close this issue for now. For further discussion this is a good candidate for a question on the https://esp32.com forums.

closed 2017-10-05 by projectgus