espressif/esp-idf #1073: SPIFFS breaks my build

forthlightning, 2017-10-03:
Hardware:

Board:	Custom board
Running arduino-esp32 as an idf component. Both updated as of this writing

Build fails before it gets to my code:

```
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'bool fs::SPIFFSFS::begin(bool, const char*, uint8_t)':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:33:31: error: 'esp_spiffs_mounted' was not declared in this scope
     if(esp_spiffs_mounted(NULL)){
                               ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:38:5: error: 'esp_vfs_spiffs_conf_t' was not declared in this scope
     esp_vfs_spiffs_conf_t conf = {
     ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:45:46: error: 'conf' was not declared in this scope
     esp_err_t err = esp_vfs_spiffs_register(&conf);
                                              ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:45:50: error: 'esp_vfs_spiffs_register' was not declared in this scope
     esp_err_t err = esp_vfs_spiffs_register(&conf);
                                                  ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'void fs::SPIFFSFS::end()':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:56:31: error: 'esp_spiffs_mounted' was not declared in this scope
     if(esp_spiffs_mounted(NULL)){
                               ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:57:55: error: 'esp_vfs_spiffs_unregister' was not declared in this scope
         esp_err_t err = esp_vfs_spiffs_unregister(NULL);
                                                       ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'bool fs::SPIFFSFS::format()':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:68:43: error: 'esp_spiffs_format' was not declared in this scope
     esp_err_t err = esp_spiffs_format(NULL);
                                           ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'size_t fs::SPIFFSFS::totalBytes()':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:79:43: error: 'esp_spiffs_info' was not declared in this scope
     if(esp_spiffs_info(NULL, &total, &used)){
                                           ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'size_t fs::SPIFFSFS::usedBytes()':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:88:43: error: 'esp_spiffs_info' was not declared in this scope
     if(esp_spiffs_info(NULL, &total, &used)){
                                           ^
make[1]: *** [libraries/SPIFFS/src//SPIFFS.o] Error 1
make: *** [component-arduino-esp32-build] Error 2

```

lucazader, 2017-10-04:
A couple of things to quickly check:
1) Did you update the submodules when you pulled the latest idf?
2) Have you tried performing a make clean before calling make?

projectgus, 2017-10-04:
It looks as if esp_spiffs.h isn't being included properly when compiling SPIFFS.cpp.

Are there any other errors or warnings before the errors you posted? Sometimes an error near the beginning can lead to a cascade of errors further down that only exist because of the first error.

If the answer to the first question is "no", can you please run a "make V=1" and post the full command line printed for "xtensa-esp32-elf-g++ ... libraries/SPIFFS/SPIFFS.cpp".

forthlightning, 2017-10-10:
Output of `make flash` after `make clean` with most updated esp-idf arduino, and toolchain is:

```
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o mconf.o mconf.c
flex -L -P zconf -o zconf.lex.c zconf.l
zconf.l:255: warning, -s option given but default rule can be matched
bison -t -l -p zconf -o zconf.tab.c zconf.y
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o zconf.tab.o zconf.tab.c
lxdialog/check-lxdialog.sh -check cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD -lncurses
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o lxdialog/util.o lxdialog/util.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lncurses
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o conf.o conf.c
cc -o conf conf.o  zconf.tab.o -lncurses
GENCONFIG
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
CC build/bootloader/main/flash_qio_mode.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
esptool.py v2.1
Building partitions from /.../solsense_esp32/ss_partition.csv...
CC build/app_trace/app_trace.o
CC build/app_trace/app_trace_util.o
CC build/app_trace/host_file_io.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
CC build/app_update/esp_ota_ops.o
AR build/app_update/libapp_update.a
CC build/arduino-esp32/cores/esp32/libb64/cdecode.o
CC build/arduino-esp32/cores/esp32/libb64/cencode.o
CC build/arduino-esp32/cores/esp32/esp32-hal-adc.o
CC build/arduino-esp32/cores/esp32/esp32-hal-bt.o
CC build/arduino-esp32/cores/esp32/esp32-hal-dac.o
CC build/arduino-esp32/cores/esp32/esp32-hal-gpio.o
CC build/arduino-esp32/cores/esp32/esp32-hal-i2c.o
CC build/arduino-esp32/cores/esp32/esp32-hal-ledc.o
CC build/arduino-esp32/cores/esp32/esp32-hal-matrix.o
CC build/arduino-esp32/cores/esp32/esp32-hal-misc.o
CC build/arduino-esp32/cores/esp32/esp32-hal-sigmadelta.o
CC build/arduino-esp32/cores/esp32/esp32-hal-spi.o
CC build/arduino-esp32/cores/esp32/esp32-hal-time.o
[... 47 lines cut ...]
                                                  ^
/.../solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'void fs::SPIFFSFS::end()':
/.../solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:56:31: error: 'esp_spiffs_mounted' was not declared in this scope
     if(esp_spiffs_mounted(NULL)){
                               ^
/.../solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:57:55: error: 'esp_vfs_spiffs_unregister' was not declared in this scope
         esp_err_t err = esp_vfs_spiffs_unregister(NULL);
                                                       ^
/.../solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'bool fs::SPIFFSFS::format()':
/.../solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:68:43: error: 'esp_spiffs_format' was not declared in this scope
     esp_err_t err = esp_spiffs_format(NULL);
                                           ^
/.../solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'size_t fs::SPIFFSFS::totalBytes()':
/.../solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:79:43: error: 'esp_spiffs_info' was not declared in this scope
     if(esp_spiffs_info(NULL, &total, &used)){
[... cut ...]
/.../solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'size_t fs::SPIFFSFS::usedBytes()':
/.../solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:88:43: error: 'esp_spiffs_info' was not declared in this scope
     if(esp_spiffs_info(NULL, &total, &used)){
                                           ^
make[1]: *** [libraries/SPIFFS/src//SPIFFS.o] Error 1
make: *** [component-arduino-esp32-build] Error 2
```

I will try `make V=1` shortly and send results.

forthlightning, 2017-10-10:
Also, when I do a `make clean`

I get the following 2 lines, may be of interest:

```
/.../esp-idf/components/bootloader/Makefile.projbuild:123: warning: undefined variable `SECURE_BOOTLOADER_KEY'
/.../esp-idf/components/bootloader/Makefile.projbuild:123: warning: undefined variable `BOOTLOADER_DIGEST_BIN'
```

forthlightning, 2017-10-10:
This is output of make V=1

xtensa-esp32-elf-c++ -std=gnu++11 -fno-rtti -Og -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -fno-rtti -DESP_PLATFORM -D IDF_VER=\"de750e9\" -MMD -MP   -DARDUINO=10800 -DESP32=1 -DARDUINO_ARCH_ESP32=1 -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/cores/esp32/libb64 -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/cores/esp32 -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/variants/esp32 -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/ArduinoOTA -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/ArduinoOTA/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/ArduinoOTA/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/EEPROM -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/EEPROM/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/ESP32 -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/ESP32/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/ESP32/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/ESPmDNS -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/ESPmDNS/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/ESPmDNS/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/FS -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/FS/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/HTTPClient -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/HTTPClient/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/HTTPClient/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/Preferences -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/Preferences/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/Preferences/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SD -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SD/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SD/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SD_MMC -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SD_MMC/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SD_MMC/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPI -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPI/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SimpleBLE -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SimpleBLE/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SimpleBLE/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/Update -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/Update/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/Update/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/WiFi -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/WiFi/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/WiFi/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/WiFiClientSecure -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/WiFiClientSecure/examples -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/WiFiClientSecure/src -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/Wire -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/Wire/src -I/Users/{personalstuff}/s [... cut ...] se32/Firmware/esp-idf/components/wpa_supplicant/port/include -I/Users/{personalstuff}/solsense32/Firmware/esp-idf/components/wpa_supplicant/../esp32/include -I/Users/{personalstuff}/solsense32/Firmware/esp-idf/components/xtensa-debug-module/include -I/Users/{personalstuff}/solsense32/Firmware/solsense_esp32/build/include  -Ilibraries/SPIFFS/src/ -c /Users/{personalstuff}/solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp -o libraries/SPIFFS/src//SPIFFS.o

forthlightning, 2017-10-20:
D:

forthlightning, 2017-10-20:
oops this was me. already had a component called spiffs from before esp-idf incorporated such a thing.

closed 2017-10-20 by forthlightning