espressif/esp-idf #1074: esp-idf-template causes

robbym, 2017-10-04:
I cloned the esp-idf-template.
Change ssid on line 25. (keeping the wrong password)
```
            .ssid = "MyWifi",
```
Compiled and flashed.

ESP32 keeps failing repeatedly:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:4456
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:11756
entry 0x40078ce4
W (73) rtc_clk: Bogus XTAL frequency: 0 MHz
I (196) cpu_start: Pro cpu up.
I (196) cpu_start: Single core mode
I (197) heap_init: Initializing. RAM available for dynamic allocation:
I (200) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (206) heap_init: At 3FFC7AB8 len 00018548 (97 KiB): DRAM
I (212) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (219) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (225) heap_init: At 4008EE8C len 00011174 (68 KiB): IRAM
I (231) cpu_start: Pro cpu start user code
I (25) cpu_start: Starting scheduler on PRO CPU.
I (36) wifi: wifi firmware version: ee52423
I (36) wifi: config NVS flash: enabled
I (36) wifi: config nano formating: disabled
I (38) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (48) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (62) wifi: Init dynamic tx buffer num: 32
I (62) wifi: Init data frame dynamic rx buffer num: 32
I (66) wifi: Init management frame dynamic rx buffer num: 32
I (72) wifi: wifi driver task: 3ffd4040, prio:23, stack:4096
I (77) wifi: Init static rx buffer num: 10
I (81) wifi: Init dynamic rx buffer num: 32
I (85) wifi: Init rx ampdu len mblock:7
I (88) wifi: Init lldesc rx ampdu entry mblock:4
I (93) wifi: wifi power manager task: 0x3ffd94bc prio: 21 stack: 2560
I (175) phy: phy_version: 359.0, e79c19d, Aug 31 2017, 17:06:07, 0, 0
I (176) wifi: mode : sta (24:0a:c4:10:5c:44)
I (298) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (955) wifi: ptate: init -> auth (b0)
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400fb130  PS      : 0x00060c30  A0      : 0x800f7487  A1      : 0x3ffd1520
0x400fb130: igmp_timer at /home/robby/esp/esp-idf/components/lwip/core/timers.c:428 (discriminator 1)

A2      : 0x3ffc5b84  A3      : 0x3ffd1540  A4      : 0x400fb130  A5      : 0x00000000
0x400fb130: igmp_timer at /home/robby/esp/esp-idf/components/lwip/core/timers.c:428 (discriminator 1)

A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800fb1d0  A9      : 0x3ffd1500
A10     : 0x00000000  A11     : 0x4008268c  A12     : 0x00000000  A13     : 0x000000f6
0x4008268c: _free_r at /home/robby/esp/esp-idf/components/newlib/./syscalls.c:41

A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000

Backtrace: 0x400fb130:0x3ffd1520 0x400f7484:0x3ffd1540
0x400fb130: igmp_timer at /home/robby/esp/esp-idf/components/lwip/core/timers.c:428 (discriminator 1)

0x400f7484: tcpip_thread at /home/robby/esp/esp-idf/components/lwip/api/tcpip.c:474

Rebooting...
```

sdkconfig is pretty much default except for brown-out detection is disabled.

projectgus, 2017-10-04:
This kind of IllegalInstructionException is often caused by the flash chip browning out and reading invalid instruction data to the CPU.

> sdkconfig is pretty much default except for brown-out detection is disabled.

Did you disable brown-out detection because it was triggering? If so, this is the root cause of the problem (unstable power).

robbym, 2017-10-04:
Hello @projectgus,

I checked what regulator we are using, and its only 300mA capable. The ESP32 apparently needs at least 500mA. Woops!

Will swap it out with something more capable and close issue if that fixes the problem.

FayeY, 2017-10-16:
Any update? Is this issue solved now?

robbym, 2017-10-16:
Yes, providing it with a more capable source of current fixed the problem.

closed 2017-10-16 by FayeY