espressif/esp-idf #1075: Uart is dropping characters when writing and sometimes overruns and resends up to 1K of previously sent data.

dmcnaugh, 2017-10-04:
Toolchain: xtensa-esp32-elf-osx-1.22.0-73-ge28a011-5.2.0.tar.gz
Platform: OSX
Esp-idf: commit e6afe28bafe5db5ab79fae213f2e8e1ccd9f937c
ESP32: DFRobot FireBeetle https://www.dfrobot.com/product-1590.html

When continuously writing **single characters  to `stdout` in a short loop**, at unpredictable intervals I am experiencing either:
* dropped characters, often around 120 consecutive characters are dropped
* some kind of buffer overrun/loopback resulting in the previous 900-1024 characters being retransmitted

I can easily reproduce the dropped character result, and the same thing occurs if I
* use the uart driver or use non-blocking mode
* use the uart driver in non-blocking mode (using `fcntl(...O_NONBLOCK...)` )
* enable or disable software flow control
(hardware flow control is not an option as I am using a the USB<->serial interface on my dev board

I have not successfully written an isolated test case that reproduces the overrun/loopback, but it happens every time I run my project code and will occur 3-6 times during writing around 45Kb I am using as a test case.

projectgus, 2017-10-04:
Hi @dmcnaugh ,

Thanks for the detailed bug report.

> I have not successfully written an isolated test case that reproduces the overrun/loopback,

If a simple test case won't trigger this, is there anything else that you can tell us about your project that may help us reproduce this? ie what peripherals are in use, what kind of load, etc?

dmcnaugh, 2017-10-05:
The DFRobot FireBeetle is Revision 0 silicon, so I wondered if the known hardware bug identified by 
```
Section 3.3 When the CPU accesses peripherals and writes one address
repeatedly, random data loss occurs.
```
might apply. But I've looked through the source and see that in `...soc/uart_reg.h`
```
#define REG_UART_AHB_BASE(i)  (0x60000000 + (i) * 0x10000 + ( (i) > 1 ? 0xe000 : 0 ) )
#define UART_FIFO_AHB_REG(i)  (REG_UART_AHB_BASE(i) + 0x0)
```
so unless this bug persists despite the recommended workaround then we can rule this out.

My application is reading a file from an sdcard using the VFS FAT via sdmmc_host which is configured in 4wire mode based on the SD Card example code using:
```
        sdmmc_host_t host = SDMMC_HOST_DEFAULT();
        // To use 1-line SD mode, uncomment the following line:
        // host.flags = SDMMC_HOST_FLAG_1BIT;
        sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
```
With the Uart I am effectively writing an ASCII based HEX dump of the file, while polling for input on the Uart in non-blocking mode with the following code:
```
	int ch = fgetc(stdin);
	if(ch != EOF) {
		ungetc(ch, stdin);
		status = RX_AVAILABLE;
	}
```
All the documentation suggests that my writes should be blocking while the FIFO is full, so I am not expecting to have to do anything other than
```
        fputc(ch, stdout);
```
to write one character at a time to the Uart.
I am also using the SPI master peripheral on VSPI_HOST as follows:
```
#define PIN_NUM_MISO 19
#define PIN_NUM_MOSI 23
#define PIN_NUM_CLK  18
#define PIN_NUM_CS   -1
spi_bus_config_t buscfg={
	.miso_io_num=PIN_NUM_MISO,
	.mosi_io_num=PIN_NUM_MOSI,
	.sclk_io_num=PIN_NUM_CLK,
	.quadwp_io_num=-1,
	.quadhd_io_num=-1
};
    ESP_ERROR_CHECK( spi_bus_initialize(VSPI_HOST, &buscfg, 1) );
```
Along with 8 x GPIOs as follows:
```
#define PIN_NUM_LED_LAT	21
#define PIN_NUM_SW_INT	22
#define PIN_NUM_SW_LAT	16
#define PIN_NUM_SW_OE	17
	gpio_set_direction(PIN_NUM_LED_LAT, GPIO_MODE_OUTPUT);
	gpio_set_direction(PIN_NUM_SW_LAT, GPIO_MODE_OUTPUT);
	gpio_set_direction(PIN_NUM_SW_OE, GPIO_MODE_OUTPUT);
	gpio_set_direction(PIN_NUM_SW_INT, GPIO_MODE_INPUT);
#define PIN_NUM_DISKA	25
#define PIN_NUM_DISKB	26
#define PIN_NUM_DISKC	27
#define PIN_NUM_DISKD	5
	gpio_set_direction(PIN_NUM_DISKA, GPIO_MODE_OUTPUT);
	gpio_set_direction(PIN_NUM_DISKB, GPIO_MODE_OUTPUT);
	gpio_set_direction(PIN_NUM_DISKC, GPIO_MODE_OUTPUT);
	gpio_set_direction(PIN_NUM_DISKD, GPIO_MODE_OUTPUT);
```
And a hardware timer on TIMER_GROUP_1, TIMER_0.
```
   timer_config_t config = {
            .alarm_en = true,
            .counter_en = false,
            .intr_type = TIMER_INTR_LEVEL,
            .counter_dir = TIMER_COUNT_UP,
            .auto_reload = true,
            .divider = 8000   /* 8000 * 12.5 = 100000 ns per tick : 10000Hz */
    };
    
    timer_init(TIMER_GROUP_1, TIMER_0, &config);
    timer_set_counter_value(TIMER_GROUP_1, TIMER_0, 0);
    timer_set_alarm_value(TIMER_GROUP_1, TIMER_0, timer_period_us);
    timer_enable_intr(TIMER_GROUP_1, TIMER_0);
    timer_isr_register(TIMER_GROUP_1, TIMER_0, &timeout_isr, NULL, 0, &timer_handle);

    timer_start(TIMER_GROUP_1, TIMER_0);
```
I think that is everything.

dmcnaugh, 2017-10-06:
Hi, I think you can close this issue now.
I have the following resolution:
As noted above I am using the DFRobot FireBeetle. This board does not use the CP2102 for USB-Serial but instead uses a STM32F042F6P6 and must be running some firmware to perform the USB-Serial conversion as this chip is a general purpose microcontroller, not a dedicated protocol converter.
I thought to eliminate this element by using an off-the-self USB-Serial board based on the FT232RL connected directly to the ESP32 Uart0 TX/RX (GPIO1/3) pins.

Using this interface I get no dropped characters and I get no buffer overrun/loopback as described above.

I must conclude there is a bug in the DFRobot firmware on the STM32F042F6P6.
I will raise this with DFRobot directly.

Sorry to waste your time.

closed 2017-10-06 by dmcnaugh

G6EJD, 2017-10-27:
Have you tried slowing down the serial baud rate to 115200?

TiSiAi, 2018-07-04:
Hello
Same problem on OS X 10.12.6
Upload sometimes only 
OK on windows same module
To Upload on OSX, I use a 3,3V serial adapter and i Force GPIO to 0V before restet and after reset I remove it. Its the procedure to program on ESP32

doronby, 2018-12-05:
Seems the same or similar problem never got resolved: https://github.com/espressif/esp-idf/issues/1075

G6EJD, 2018-12-05:
It’s a problem with the DFRobot firebeettle board UART implementation.