espressif/esp-idf #1076: SYSTEM_EVENT_STA_DISCONNECTED not fired

kglowacki, 2017-10-04:
I have a simple program that connects to wifi and then I listen to wifi event handlers.
Scenario:
1. I start the program and it connects to existing wifi network, following events are triggered: -
- SYSTEM_EVENT_STA_START
- SYSTEM_EVENT_STA_CONNECTED
- SYSTEM_EVENT_STA_GOT_IP
2. while program is in a main loop (sleeping), I turn wifi hotspot off.
3. no wifi event is generated (I'd expect SYSTEM_EVENT_STA_DISCONNECTED), instead of I see "wifi: bcn_timout,ap_probe_send_start" message being printed over and over
4. when I turn wifi hotspot on, handler receives following events:
- SYSTEM_EVENT_STA_DISCONNECTED
- SYSTEM_EVENT_STA_CONNECTED
- SYSTEM_EVENT_STA_GOT_IP

projectgus, 2017-10-10:
Hi @kglowacki ,

Thanks for reporting this. Are you able to post the code that you're using, or a log with wifi logging enabled, so we can more easily reproduce?

Are you using Mongoose or ESP-IDF directly? Which version(s)?

liuzfesp, 2017-10-11:
@kglowacki it's expected behavior.
When the soft-AP is shutdown, the soft-AP no longer send beacon per beacon interval.
If the STA lost >=60 continuous beacons, it print “bcn timeout” log. 
Then STA try to probe the AP by sending a probe request, if no probe response is from AP, STA continue sends probe request.
After 5 probe request, if still no probe response. STA generates SYSTEM_EVENT_STA_DISCONNECTED.

closed 2017-10-11 by FayeY

pedro-n-rocha, 2018-04-11:
`After 5 probe request, if still no probe response. STA generates SYSTEM_EVENT_STA_DISCONNECTED.`

I am not seeing this behaviour SYSTEM_EVENT_STA_DISCONNECTED never gets fired.

@liuzfesp The op description is acurate , how is this expected behaviour ?

pedro-n-rocha, 2018-04-11:
ok , debbugged this down to this : 

I was doing wifi scans on another task , it seems  if a scan initiates before or during bcn_timout,ap_probe_send_start procedures it just  prevents it from continuing and triggering the DISCONNECTED event , somethig for you guys to consider .