espressif/esp-idf #1079: BLE GATT with multiple identical uuid

fchaxel, 2017-10-05:
Hi,

I'm testing BLE GATT, and I'm facing a problem.

Into a service one can have more than one characteristic with the SAME uuid (for instance two Analog_Input). It's accepted and working fine with the ESP32 stack.

But the event ESP_GATTS_ADD_CHAR_EVT and the data within the structure provided -esp_ble_gatts_cb_param_t- is not suffisant to distinguish between the first, then 2nd, ... differents characteristics with the same Id. So to obtain the attr_handle the event data are not enought.

Of course since the attr_handle given by the ESP BLE stack is always the same for a given configuration ... some 'working but not good' code can be used to go ahead.

I'm right ?

lucazader, 2017-10-08:
As a general rule, You should not use two characteristics with the same UUID in the same service in BLE. It can and will cause lots of issues, like the one mentioned above both internally in the esp32 and with connected central applications.

If you really need to use the same characteristic UUID, then perhaps use a difference service to house the characteristics with the same UUID.

heyinling, 2017-10-08:
@fchaxel `esp_ble_gatts_create_attr_tab` is preferred to create gatt service. it will return an array of handles, the same order with the table of attributes you created. identical UUID chars will have different handles, you can access with handles then.

if you use old add char API, the event won't give you enough info. need to add extra flag yourself.

fchaxel, 2017-10-09:
Hi,

No, multiple characteristic UUID are acceptable. In fact why a service could'nt be used to gives access to 3 separeted LED, or 10 buttons for instance ? Of course we can put all into an unique charac., but if it's a standard UUID (for instance 0x2A59 for one and only one Analog Output) it's not possible without loosing genericity.

A service may have multiple characteristic definitions with the same Characteristic UUID (Bluetooth 4, Core specification, page 537)
https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=229737

I will try esp_ble_gatts_create_attr_tab

Thank's to all.

closed 2017-10-09 by fchaxel

ManniAT, 2017-11-11:
If you have a look at https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.environmental_sensing.xml the use of the same UUID is documented.
 For environmental you have to add a ES_MEASUREMENT description when using the same UUID - but doing this is a common thing (search for if_multiple_instances_of_same_characteristic_optional_otherwise).
Imagine a weather station - you have a temperature from the outdoor sensor and on from indoor (same for humidity).