espressif/esp-idf #1080: SPI Master - second device (spi_bus_add_device) does not work

OtherCrashOverride, 2017-10-05:
When using spi_bus_add_device to add two devices to a SPI bus with different CS lines, the second device does not operate properly if one is half duplex and the other full duplex.  If both are full duplex, the issue does not occur.

Spritetm, 2017-10-06:
Can you expand a bit on 'does not operate properly'? What symptoms do you notice?

OtherCrashOverride, 2017-10-06:
Due to time constraints, I did not have the opportunity to explore what exactly "does not operate properly" entails.

The configuration being tested is an ILI9341 SPI display as the first SPI device and a XPT2046 touch panel interface as the second SPI device.  The first device is configured in half duplex mode while the second is full duplex.  When either is the only device added (spi_bus_add_device), correct operation is observed.  When the display is added followed by the touch panel, the touch panel does not report correct data.  Changing the display device to full duplex yields correct operation of both devices.

ginkgm, 2021-05-06:
Hi @OtherCrashOverride ,

Sorry for long long no-reply ....

The issue was reported quite long ago, during which we have introduced a similar issue on 4.0 and fix it on master (see: https://github.com/espressif/esp-idf/issues/4641). (The fix backporting to 4.0, 4.1 and 4.2 are in progress....)

Please help test again, or wait for patches to 4.0-4.2. (they will also close this issue upon merged)

Alvin1Zhang, 2021-09-02:
Thanks for reporting and sorry for slow turnaround. 

- Fix on release/4.0 is https://github.com/espressif/esp-idf/commit/9e9ebed8612a36ea334ecd5a23d20c71afa72da2
- Fix on release/4.1 is https://github.com/espressif/esp-idf/commit/9a8dbbc281ce642868b4b3f701525cf92458f0b3
- Fix on release/4.2 is https://github.com/espressif/esp-idf/commit/e3fd4b097b350403a8dd1ec7cdc0f40b7f9aeedb

Feel free to reopen.

closed 2021-09-02 by Alvin1Zhang