espressif/esp-idf #1081: Local SDK configuration

seppestas, 2017-10-06:
When working with multiple people on a project that uses IDF-SDK it would be very handy to have a way to overwrite SKD settings that are environment specific locally.

E.g setting like `CONFIG_PYTHON` and `CONFIG_ESPTOOLPY_PORT` are very environment specific. Being able to change them without having to change the sdkconfig file would be very interesting since this would allow the sdkconfig file to be checked in to version control, allowing other important settings to be shared, while keeping environment specific settings local.

Is there a way this can be done in the current system? If not, I propose some kind of sdkconfig.local file that can be used to overwrite settings locally. This file could than be excluded from the version control system making sure it remains local. If the file is not present, the default configuration from the sdkconfig file is used.

tuanpmt, 2017-10-06:
I usually use environment variables
For example, `main/component.mk`
```
ifeq ($(MODE),master)
	CFLAGS += -DIS_MASTER
endif

#force re-compile main.c
main.o: .FORCE
.FORCE:
```
`main/main.c`
```
#ifdef IS_MASTER
#warning "Your device will working as MASTER"
#else
#warning "Your device will working as SLAVE"
#endif
```
And 
`MODE=master make flash monitor`

seppestas, 2017-10-06:
Is it possible to pass these variables to the ESP tools? The variables I talked about (`CONFIG_PYTHON` and `CONFIG_ESPTOOLPY_PORT`) are not only used by the preprocessor, they are also used by e.g the monitor and ESPTOOL.

tuanpmt, 2017-10-06:
with COM port, you can use `ESPPORT=COM1 make flash`. I have not tried with python

FayeY, 2017-10-11:
Hi @seppestas , did the above suggestion help you resolve the issue?

projectgus, 2017-10-11:
Unfortunately only the ESPPORT & ESPBAUD variables can override sdkconfig settings in this way.

We've discussed something like this (local non-project-specific config) in the past but so far we haven't implemented anything yet, so keeping this open as an enhancement request is valid.

seppestas, 2017-11-08:
The `ESPPORT` variable works great for setting the serial port of the ESP. Now I mainly want to find a way to make the "Python 2 interpreter" setting a bit more portable.

Could this be solved by using an environmental variable for the setting that defaults to `python`? How could this be done?

projectgus, 2019-03-01:
Hi @seppestas ,

Sorry noone got back to you earlier about this. Two important things have changed in this regard:

* We now support Python 2 and 3, so it's more likely the system default Python will work with ESP-IDF.

* For the [new CMake-based build system](https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/), we've moved all 3 of these configuration settings (python interpreter, serial port, serial baud) out of KConfig entirely.

  - If you use the `idf.py` command line tool then Python is inferred to be the interpreter that runs idf.py, and the other two are provided as command line arguments (with defaults from environment variables ESPPORT & ESPBAUD, same as before).
  - If you run CMake directly then the environment variables PYTHON, ESPPORT and ESPBAUD can be used to set these 3 parameters. Or python will default to "python" if the variable is unset.

closed 2019-03-01 by projectgus