espressif/esp-idf #1082: Bluetooth Controller: HCI Set Connection Encryption fails unexpectedly

mringwal, 2017-10-06:
Hi there. I got a BTstack user that fails to connect via Classic to a SEPURA STP-9000 TETRA NETWORK. Although pairing succeeds and link keys are generated/provided, the Set Connection Encryption fails.

Here's my log analysis of https://github.com/bluekitchen/btstack/issues/81#issuecomment-334219796 

While the Remote Supported Features indicate support for Secure Simple Pairing, legacy pairing with PIN is used successfully. In the single log, the connection stops at this point.

On reconnection attempts, the ESP32 Controller asks for the link key before the Connection Complete event is received. Later, when BTstack sends HCI Authentication Requested, the Controller asks again for the link key, which is a bit suspicious, but we get a Authentication Complete (Success) event. The next HCI Set Connection Encryption command however fails with "Invalid HCI Command Parameters". Interestingly, there's also a Command Status Event for "Refresh Encryption Key" although no HCI Refresh Encryption Key Command was sent before.

Why is there a second link key request for the same connection? Could this be a problem?
Is the Link Key Request before the Connection Complete Event causing issues?

A shortened & commented log here:
..
[00:02:24.265] EVT <= 04 0A 4F 65 01 96 1E 00 08 02 00 01
[00:02:24.265] LOG -- hci.c.1859: Connection_incoming: 00:1E:96:01:65:4F, type 1
[00:02:24.266] LOG -- hci.c.181: create_connection_for_addr 00:1E:96:01:65:4F, type ff
[00:02:24.274] LOG -- hci.c.3103: sending hci_accept_connection_request, remote eSCO 0
[00:02:24.281] CMD => 09 04 07 4F 65 01 96 1E 00 01
[00:02:24.286] EVT <= 0F 04 00 05 09 04
[00:02:24.290] EVT <= 6E 00
[00:02:24.324] EVT <= 17 06 4F 65 01 96 1E 00

First Link Key Request

[00:02:24.325] LOG -- hci.c.1969: HCI_EVENT_LINK_KEY_REQUEST
[00:02:24.325] LOG -- hci.c.3131: responding to link key request
[00:02:24.480] CMD => 0B 04 16 4F 65 01 96 1E 00 5B B3 5E 0B CD 8B 84 14 2C 9E F4 8C 2B 9C 84 0A
[00:02:24.489] EVT <= 0E 0A 05 0B 04 00 4F 65 01 96 1E 00
[00:02:24.493] EVT <= 6E 00
[00:02:24.546] EVT <= 03 0B 00 80 00 4F 65 01 96 1E 00 01 01

Connection Complete

[00:02:24.546] LOG -- hci.c.1883: Connection_complete (status=0) 00:1E:96:01:65:4F
[00:02:24.548] LOG -- hci.c.1896: New connection: handle 128, 00:1E:96:01:65:4F
[00:02:24.555] LOG -- hci.c.3719: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:02:24.561] EVT <= 61 01 01
[00:02:24.564] CMD => 1B 04 02 80 00
[00:02:24.568] EVT <= 1B 03 80 00 05
[00:02:24.571] EVT <= 0F 04 00 05 1B 04
[00:02:24.575] EVT <= 0B 0B 00 80 00 FF FE 8F F8 1B 18 00 80
[00:02:24.580] LOG -- hci.c.1962: HCI_EVENT_READ_REMOTE_SUPPORTED_FEATURES_COMPLETE, bonding flags 2, eSCO 1
..
[00:02:25.139] LOG -- hci.c.3852: gap_request_security_level 2, current level 0
[00:02:25.141] CMD => 11 04 02 80 00

Authentication Requested

[00:02:25.144] EVT <= 0F 04 00 05 11 04

Second Link Key Request

[00:02:25.148] EVT <= 17 06 4F 65 01 96 1E 00
[00:02:25.152] LOG -- hci.c.1969: HCI_EVENT_LINK_KEY_REQUEST
[00:02:25.157] EVT <= 6E 00
[00:02:25.160] ACL => 80 00 10 00 0C 00 01 00 03 04 08 00 41 00 60 00 01 00 00 00
[00:02:25.167] EVT <= 6E 00
[00:02:25.170] LOG -- hci.c.3131: responding to link key request
[00:02:25.326] CMD => 0B 04 16 4F 65 01 96 1E 00 5B B3 5E 0B CD 8B 84 14 2C 9E F4 8C 2B 9C 84 0A
[00:02:25.335] EVT <= 13 05 01 80 00 01 00
[00:02:25.339] EVT <= 0E 0A 05 0B 04 00 4F 65 01 96 1E 00
[00:02:25.344] EVT <= 6E 00
[00:02:25.350] EVT <= 06 03 00 80 00

Authentication Complete (Success)

[00:02:25.350] CMD => 13 04 03 80 00 01

Set Connection Encryption = 1

[00:02:25.354] EVT <= 0F 04 0C 05 53 0C

Command Status Refresh Encryption Key (Command Disallowed)

[00:02:25.357] EVT <= 0F 04 12 05 13 04

Command Status Set Connection Encyrption (Invalid HCI parameters)

flograsso, 2017-12-06:
How can i deactivate the "EVT , LOG, ACL and CMD" messages on the Serial port?

b3kz0d, 2017-12-12:
> How can i deactivate the "EVT , LOG, ACL and CMD" messages on the Serial port?
Hi, open main.c file in esp/esp-idf/components/btstack folder and find function called hci_dump_open and comment it. That is all.

blueMoodBHD, 2019-04-28:
This issue may have been fixed, refer to commit e7f85f1987aa9479c2dbab638ca83bcaef99be00

mringwal, 2019-04-28:
Added a comment here: https://github.com/bluekitchen/btstack/issues/81
If we get a positive response, or none at all in the next days, I guess we can close it here (as I wasn't able to reproduce it).

mringwal, 2019-05-06:
Closing it for now as I was never able to reproduce it.

closed 2019-05-06 by mringwal