espressif/esp-idf #1084: Random crash when receiving a continuous stream on socket.

karawin, 2017-10-06:
My software runs pretty well but time to time this bug  happens in a random time.
I don't have timers in it apart for the sockets.
The software receives a stream of mp3 radio.

(gdb) bt
#0  0x400896d6 in invoke_abort ()
    at C:/msys32/home/jp/esp/esp-idf/components/esp32/panic.c:139
#1  0x40089a9c in _esp_error_check_failed (rc=259,
    file=0x3f41f914 "C:/msys32/home/jp/esp/esp-idf/components/esp32/ets_timer_legacy.c", line=89, function=0x3f41f9e0 <__func__$5258> "ets_timer_arm",
    expression=0x3f41f974 "esp_timer_start_once(ESP_TIMER(ptimer), time_us)")
    at C:/msys32/home/jp/esp/esp-idf/components/esp32/panic.c:565
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Seems to come from:
#1  0x400dfbb5 in sta_input ()
No symbol table info available.
#2  0x400e02b4 in sta_rx_cb ()
No symbol table info available.
#3  0x400e833d in ppRxPkt ()
No symbol table info available.
#4  0x400e9bf2 in ppTask ()
No symbol table info available.

But cannot investigate further.

kglowacki, 2017-10-07:
I have very similar problem, it happens after some time of network activity (a few hours):

ESP_ERROR_CHECK failed: esp_err_t 0x103 at 0x4012a9a9
ESC[0;33m0x4012a9a9: ets_timer_arm at /opt/esp/esp-idf/components/esp32/./ets_timer_legacy.c:88 (discriminator 1)
ESC[0m

Backtrace: 0x4008781c:0x3ffcb390 0x40087b99:0x3ffcb3b0 0x4012a9a9:0x3ffcb3d0 0x400e6589:0x3ffcb3f0 0x400e67ba:0x3ffcb420
ESC[0;33m0x4008781c: invoke_abort at /opt/esp/esp-idf/components/esp32/./panic.c:553
ESC[0m
ESC[0;33m0x40087b99: _esp_error_check_failed at /opt/esp/esp-idf/components/esp32/./panic.c:565
ESC[0m
ESC[0;33m0x4012a9a9: ets_timer_arm at /opt/esp/esp-idf/components/esp32/./ets_timer_legacy.c:88 (discriminator 1)
ESC[0m
ESC[0;33m0x400e6589: pm_rx_beacon_process at ??:?
ESC[0m
ESC[0;33m0x400e67ba: pm_rx_beacon_process at ??:?

I have plenty of heap free and none of tasks run out of stack. I think it started occurring after updating from 1.0 to 2.1.

karawin, 2017-10-10:
Seems to be a stack corruption. Closing

karawin, 2017-10-12:
Reopened. It appears time to time.

karawin, 2017-10-18:
**Changed in ets_timer_legacy.c**
void ets_timer_arm(ETSTimer *ptimer, uint32_t time_ms, bool repeat_flag)
{
    uint64_t time_us = 1000LL * (uint64_t) time_ms;
    assert(timer_initialized(ptimer));
    esp_timer_stop(ESP_TIMER(ptimer));  // no error check
    if (!repeat_flag) {
        ESP_ERROR_CHECK( esp_timer_start_once(ESP_TIMER(ptimer), time_us) );
    } else {
        ESP_ERROR_CHECK( esp_timer_start_periodic(ESP_TIMER(ptimer), time_us) );
    }
}

**to** 
void ets_timer_arm(ETSTimer *ptimer, uint32_t time_ms, bool repeat_flag)
{
    uint64_t time_us = 1000LL * (uint64_t) time_ms;
    assert(timer_initialized(ptimer));
	esp_err_t err;
    esp_timer_stop(ESP_TIMER(ptimer));  // no error check
    if (!repeat_flag) {
//        ESP_ERROR_CHECK( esp_timer_start_once(ESP_TIMER(ptimer), time_us) );
        err =  esp_timer_start_once(ESP_TIMER(ptimer), time_us) ;
		if (err != ESP_OK) 
		{
			printf("err: %d, timer: %x, time_us:%Ld, time_ms:%d\n",err,(int)ptimer,time_us, time_ms);
			abort();
		}
    } else {
        ESP_ERROR_CHECK( esp_timer_start_periodic(ESP_TIMER(ptimer), time_us) );
    }
}

After a random time of running the application i see with gdb:
#0  0x400898e2 in invoke_abort ()
    at C:/msys32/home/jp/esp/esp-idf/components/esp32/panic.c:139
#1  0x400899ae in abort ()
    at C:/msys32/home/jp/esp/esp-idf/components/esp32/panic.c:148
#2  0x4012de01 in ets_timer_arm (ptimer=0x3ffb4f48, time_ms=240,
    repeat_flag=false)
    at C:/msys32/home/jp/esp/esp-idf/components/esp32/ets_timer_legacy.c:94
#3  0x400e76b0 in pm_rx_beacon_process ()
#4  0x400e78e2 in ?? ()

With the log:
err: 259, timer: 3ffb4f48, time_us:0, time_ms:240000
abort() was called at PC 0x4012ddfe on core 1

**That confirm the diagnostic of kglowacki.
It seems that the uint32_t time_ms parameter is wrong: 240000 in place of 240.**

karawin, 2017-10-20:
Seen to be not reproductible with the last idf. Closing

closed 2017-10-20 by karawin