espressif/esp-idf #1085: Enhancement: Implement a shim/function for access(2).

nkolban, 2017-10-07:
The ESP-IDF provides a wealth of Posix implementation functions and for this the community is grateful.  Recently, we came across a system call that does not appear to be implemented.  That system call is `access(2)`.  See:

https://linux.die.net/man/2/access

The request of this issue is to ask that access(2) be supported/provided by the ESP-IDF.  I'm not going to try and recommend an implementation.  Some guesses that in my ignorance I may come up with include:

* A shim based of stat() that merely checks for the existence of the file and ignores the access mode.
* Extension of the VFS specification to include a file system dependent implementation of access()

FayeY, 2017-11-03:
Fixed in [ab13db0](https://github.com/espressif/esp-idf/commit/ab13db0e452b8de3b7abccbcfb6d0ba1505803af).

igrr, 2017-11-03:
My bad, I closed the wrong issue; meant to close TW15698. This one is still to-do.

closed 2018-05-18 by igrr