espressif/esp-idf #1088: E (2894) BT: config parse returned with error code: 2

jstar88, 2017-10-08:
As title, i got the bold text below in the monitor.
Any suggestions?
`  
D (1714) BluetoothManager: esp_bt_controller_enable
D (1714) phy_init: loading PHY init data from application binary
D (1714) nvs: nvs_open_from_partition phy 0
D (1714) phy_init: esp_phy_load_cal_data_from_nvs: failed to open NVS namespace (0x1102)
W (1724) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (1884) phy: phy_version: 359.0, e79c19d, Aug 31 2017, 17:06:07, 0, 2
D (1884) nvs: nvs_open_from_partition phy 1
D (1884) nvs: nvs_set cal_version 4 359
D (1884) nvs: nvs_set_blob cal_mac 6
D (1884) nvs: nvs_set_blob cal_data 1904
D (1894) nvs: nvs_close 1
D (2894) BluetoothManager: esp_bluedroid_init
D (2894) nvs: nvs_open_from_partition bt_config.conf 1
D (2894) nvs: nvs_get_str_or_blob bt_cfg_key
**E (2894) BT: config_parse returned with err code: 2**

D (2894) nvs: nvs_close 2
D (2904) nvs: nvs_open_from_partition bt_config.conf 1
D (2904) nvs: nvs_set_blob bt_cfg_key 0
D (2914) nvs: nvs_close 3 
`

Yulong-espressif, 2017-10-11:
@jstar88 you should use the nvs_flash_init function before init the bt stack, thanks.

jstar88, 2017-10-12:
@Yulong-espressif  in my code I use nvs_flash_init as first in main function.
Then i create a task to manage bluetooth and another to manage GAP.
The error seems appear randomically at power on and i noticed that when it appear, then BLE client stop to retrive the avertising from remote sensors after 1-2 hours. ( no warning-debug-info messages are printed)
Viceversa, if the error is not shown, then the BLE keep working fine for days

igrr, 2017-10-12:
When you call nvs_flash_init, do you check the error code? If not, can you add a check for the error code there and see if that is anything other than ESP_OK when the error happens?
Also, what kind of hardware are you using (board/module)?

jstar88, 2017-10-15:
yes,  i'm using the code from examples to init nvs with the dev board. 
Could it be related to the partition table?
I was using a custom partition of type "data" to store things, and seems like the error disappear when i changed the type to 0x41.

Yulong-espressif, 2017-12-15:
@jstar88 Do you fix this issue now?

jstar88, 2017-12-17:
yes

closed 2017-12-21 by FayeY