espressif/esp-idf #1089: fwrite broken

ctag-fh-kiel, 2017-10-08:
When using fwrite on SD card write fails once file approaches 1Mbyte size, possibly an issue?

    char * buf= malloc(1024);
    bzero(buf, 1024);
    (*buf) = 'C';
    for(int i = 0; i<1024;i++){
	ESP_LOGI(TAG, "Writing block %d", i);
    	fwrite(buf, 1, 1024, f);
    }
    free(buf);
    fclose(f);

Error:
I (4902) example: Writing block 1022
I (4902) example: Writing block 1023
E (5032) sdspi_host: sdspi_host_start_command: cmd=24 error=0x107
E (5032) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (5032) diskio_sdmmc: sdmmc_write_blocks failed (263)

riataman, 2017-10-12:
This seems like a duplicate of https://github.com/espressif/esp-idf/issues/1093

ctag-fh-kiel, 2017-10-13:
I tried adapting the #define from #1093 no difference.

However, I tried using different SD cards:
- Toshiba 16GB same error
- SanDisk 16GB same error
- Intenso 8GB WORKS also without adaptation from #1093 

Is it due to SD size > 8GB?

closed 2017-11-14 by FayeY