espressif/esp-idf #1090: Clarification: Details needed on BLE advertising properties p_service_uuid and service_uuid_len

nkolban, 2017-10-08:
As we study the BLE advertising API, we find we can populate a structure called `esp_ble_adv_data_t`.  A reference to this can be found here:

http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv218esp_ble_adv_data_t

There are two properties I would like to call out ... these are:

* `uint16_t service_uuid_len`
* `uint8_t p_service_uuid`

The documentation for `p_service_uuid` seems to declare that it is an array of UUIDs.  What then does that say about `service_uuid_len`?  Should it be the count of elements in the array or the size of the array in bytes?

Since UUIDs can be either 2 bytes, 4 bytes or 16 bytes, how do we inform the caller of the "size type" of the UUIDs?   For example, if I have an array of 8 2-byte UUIDs the length would be 16 bytes while if I have an array of just 1 16-byte UUID, the array would also be 16 bytes ... but yet the advertising packet in BLE has to know what the size of the UUIDs should be and we appear to have ambiguity.

lucazader, 2017-10-08:
Hi Neil,

I think the current way that the `esp_ble_adv_data_t` is configured and handled is not suitable for your application. From my experience using the api you are only able to set 1 UUID with a size of `service_uuid_len`. So you can have 1 UUID that is either 16,32 or 128bit in size.

However you need to advertise with multiple UUIDs.
For this i believe you will need to build up your advertising data manually and set the using the `esp_ble_gap_config_adv_data_raw` and `esp_ble_gap_config_scan_rsp_data_raw` functions.
You will also need to use `ESP_BLE_AD_TYPE_16SRV_CMPL` as the advertising data type.

Below is an example of how this data would be built up using UUIDs from 0x1231-0x1238

```
uint8_t adv_data_raw[27] = {
          2, ESP_BLE_AD_TYPE_FLAG, (ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT), // advertising flags
          0x05, ESP_BLE_AD_TYPE_INT_RANGE,
          0x06, 0x00, // BLE connection settings between 7.5ms
          0x30, 0x00, // and 30ms
          17, ESP_BLE_AD_TYPE_16SRV_CMPL,
          0x31, 0x12,
          0x32, 0x12,
          0x33, 0x12,
          0x34, 0x12,
          0x35, 0x12,
          0x36, 0x12,
          0x37, 0x12,
          0x38, 0x12};
```

negativekelvin, 2017-10-09:
I think service_uuid_len should be byte length of p_service_uuid array and should always be 16*n where n is number of uuid fields because even if you have a 16bit uuid it is specified as 16 byte array and then is detected as 16-bit based on the contents of those 16 bytes.

nkolban, 2017-10-09:
@negativekelvin   Oooh!!  That would be interesting.   To recap, a 16bit and 32bit UUID can be specified in this array by converting them to 128bit UUIDs and then, when we provide them as an array parameter in the structure passed to `esp_ble_gap_config_adv_data` the logic within decodes their intent?  That would be useful!!!     I wonder what would happen if we provided a mix of 16 bit and 128 bit uuids ....?  I wonder if the logic is smart enough to build the two types of advertising types in the advert?

negativekelvin, 2017-10-09:
https://github.com/espressif/esp-idf/blob/08b756b6731581d5684bd871dee3842f8bf840d3/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c#L309

But multiple shorter uuids should work

nkolban, 2017-10-09:
Howdy,
I'm not sure what you wanted me to see in that last post.  Looking at code may be the source of truth, but unfortunately I'm too slow to read between the lines.  While I think I understand that we can only have advertising space for one 128bit UUID, I think the up-shot of an earlier post you made was that I can supply multiple 16bit or 32bit UUIDs to be advertised by converting them to their 128bit form and passing them in to the esp_ble_gap_config_adv_data() call as an array of 128bit uuids in the `p_service_uuid` parameter and the logic of esp_ble_gap_config_adv_data() will look at each one and say to itself "Hmmm ... this 128bit UUID can be represented as a 16/32bit UUID by just lopping-off the remaining/constant data and flagging it as a 16bit/32bit UUID in the advertisement".

negativekelvin, 2017-10-09:
Yes you're right, the 31byte payload could fit various combinations of uuids of different lengths but not more than one 128bit.

Yulong-espressif, 2017-10-11:
@nkolban Hi, actually the  service_uuid_len value should be the 16*n(n = 0,1,2,3,4...) multiple, even if the practical UUID size is 2 bytes, 4 bytes or 16 bytes. Because  We set all UUID types are converted into a standard 16 bytes UUID type in the application layer. We will added some comment for this type. Thank you very much.

closed 2017-10-17 by nkolban