espressif/esp-idf #1093: SD Card timing out

riataman, 2017-10-09:
Hey,

I have a project where I'm recording 44.1khz 2 channel 16bit PCM data from i2s into an SD card(about 172KB/s). If I let it record continuously it would eventually fail with an error like this:

sdmmc_host_wait_for_event returned 0x107

0x107 being ESP_ERR_TIMEOUT. I ended up fixing it by changing:

#define SDMMC_MAX_EVT_WAIT_DELAY_MS 1000

to

#define SDMMC_MAX_EVT_WAIT_DELAY_MS 1600

I found that the card latency can be that big, specially if you have an SD card that doesn't have any more free blocks in the card (has been full at some point) and it needs to go and erase one for a write.

I tested this in SPI and 1bit SD mode.

riataman, 2017-10-09:
SDMMC_MAX_EVT_WAIT_DELAY_MS has a comment stating this this is a safeguard for the card being removed so might as well make this very large (5s?)

igrr, 2017-10-09:
I think we can make this timeout configurable, would that work for you?

riataman, 2017-10-09:
That certainly would.

profatkuas, 2017-10-12:
I have a similar issue with writing large files, may I ask how large your files are when writing?
Do you use fwrite to write the binary data?

riataman, 2017-10-12:
@profatkuas: MY files can get up to about 100Mb. And yes I was using fwrite. You can try to change SDMMC_MAX_EVT_WAIT_DELAY_MS  to 2000 in the idf source code and see if it fixes that for you.

ctag-fh-kiel, 2017-10-12:
Would you mind sharing a code snippet of the SD / SPI config part as well as the part you are writing with fwrite? Thx much...

riataman, 2017-10-12:
I'm just using the default setup in the storage example. The change that I did was in idf itself:

\esp-idf\components\driver\sdmmc_transaction.c(41):#define SDMMC_MAX_EVT_WAIT_DELAY_MS 3000

Its not configurable at the moment, but seems like that might change in the future. Also it does probably needs a better default if there are more people running into this.

ctag-fh-kiel, 2017-10-13:
I tried adapting the #define 

#1089 seems to be still different, with following results using different SD cards:

- Toshiba 16GB same error as #1089 
- SanDisk 16GB same error as #1089 
- Intenso 8GB WORKS also without adaptation from #1093

Possibly #1089 is due to SD size > 8GB?

closed 2017-10-18 by igrr

igrr, 2018-07-18:
Since the commit linked above, timeout can be adjusted via `command_timeout_ms` (i.e. `card->host.command_timeout_ms`)