espressif/esp-idf #1094: UDP sockets always close with an error

dpgeorge, 2017-10-10:
It seems that calling `close(s)` on a UDP (SOCK_DGRAM) socket always returns -1, meaning an error occurred.  This happens right after creating a socket and also after sending/receiving data.  There is no such problem with SOCK_STREAM sockets.

Simple test code (placed eg in http_request example's main loop):
```c
int s = socket(AF_INET, SOCK_DGRAM, 0); 
int ret = close(s);
printf("UDP: s=%d ret=%d\n", s, ret);
```
The above code prints -1 for the ret value, whereas I would expect it to print 0.

projectgus, 2017-10-10:
Thanks for reporting this, @dpgeorge .

I think I can see the underlying problem: lwip_netconn_do_delconn() in api_msg.c doesn't set the return "err" to pass back to the the calling task when deleting a UDP socket. But tcpip_apimsg() in api_lib.c:80 sets ERR_VAL as a catch-all for API functions which don't set the error value correctly. This comes back to lwip_close() in sockets.c and results in the failure.

We will work on producing a fix.

projectgus, 2017-10-24:
This issue should be fixed as of e2d077cea19e7d7b7fd891899176363332ce9058.

closed 2017-10-24 by projectgus

dpgeorge, 2017-10-24:
Thanks @projectgus.  I will check it in due course when we update the IDF.