espressif/esp-idf #1095: Enhance esp_wifi_ap_get_sta_list / wifi_sta_info_t to include RSSI, protocol

nickzoic, 2017-10-10:
When acting as an AP, there's a call (`esp_wifi_ap_get_sta_list`) to get the list of stations connected.
This returns a list connected stations as an array of `wifi_sta_info_t`, which at the moment only has the mac address.

It'd be helpful to also make available the most recent RSSI measurement and perhaps which protocol (b,g,n,lr) the station is using.

The station RSSI is also available from the `esp_wifi_set_vendor_ie_cb` mechanism, but matching this data up to the station list is fraught. See also: https://github.com/micropython/micropython-esp32/issues/186

jakkra, 2017-10-11:
Also the country code of the AP would be helpful to have.

liuzfesp, 2017-11-26:
@nickzoic it's in our TODO list, will support it in idf3.1

nickzoic, 2017-11-26:
Brilliant, thank you!

XinDeng11, 2018-05-11:
Hi, @nickzoic, we have add the country information in the wifi_ap_record_t structure, after 3.1 release, you will found it add a country parameter in it, or you can check the commit f3df30f9 after it merged, it may need few days for review and test

XinDeng11, 2018-05-11:
Hi@nickzoic, we have added the rssi you need in the wifi_sta_info_t, you will find it after 3.1 release, if you earge to use, you can pull down the f3df30f9 after it merged, but it may need few days to merge for review and test

nickzoic, 2018-05-14:
That's great, we'll add it to MicroPython as soon as we can!

closed 2018-05-19 by igrr

nickzoic, 2018-05-24:
Brilliant, thanks Ivan!