espressif/esp-idf #1096: GPIO 36 & 39 input interrupt continually fires for high input when ADC1 is also in use

bpietsch, 2017-10-10:
I'm finding that when I use GPIO 36 or 39 as an input that when the input signal is high and ADC1 is in use on other pins that the GPIO interrupt repeatedly fires.  Disabling use of ADC1 or using other GPIO inputs (such as 37 or 38) does not cause this issue to occur.  The values read from 36 and 39 still appear to be correct, but the continual firing of the interrupt is an obvious inefficiency.

See also:
https://www.esp32.com/viewtopic.php?t=1472

costaud, 2017-10-18:
Do you have any external pull-up resistor on the GPIOs? Since there are no internal pull-ups on input-only IOs.

bpietsch, 2017-10-22:
I have an external pull-down resistor... must it be a pull-up?

closed 2018-03-12 by FayeY

mirko, 2020-09-08:
All the comments - in code - as well as section 3.11 as part of https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf only mention ADC.
As far as I can see there's no official info about GPIO36/39 /not to be used at all/ in conjunction with WiFi - no matter if ADC is in use or not.
As a matter of fact, though, when pulling GPIO36 high, it triggers interrupts constantly when WiFi (and one of its power saving modes) is enabled.
While this having hit me quite hard (PCB design and order screwed), I'd really like to know why this issue only occurs when GPIO36/GPIO39 are being pulled high - and not when left floating?

mirko, 2020-09-08:
All the comments (in code) as well as section 3.11 as part of https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf only either:
a) WiFi should not be used together with ADC
and/or
b) GPIO36/39 might toggle for RTC peripherals when powering up.

So, if I'm not using ADC I'm fine with using GPIO36/39, hm?
Unfortunately not.

And as far as I can see there's no official info about GPIO36/39 /not to be used at all/ in conjunction with WiFi - no matter if ADC is in use or not.
As a matter of fact, though, when pulling GPIO36/39 high, they toggle states and cause interrupts constantly when WiFi (and one of its power saving modes) is enabled.

This issue hit me really, really hard - and given all the docs and reports I'm aware of now, it's not covered by a single warning / errata entry. I would still not know not to do this. So either I'm overlooking some or I'd really appreciate a big fat warning stating sth. alike: "Do expect frequent (internal) state toggles (bound) interrupts on GPIO36/39 when using WiFi!"

mirko, 2020-09-08:
And why on earth do GPIO36/39 only toggle states, when being pulled high? No interrupts triggered when left floating.

terjeio, 2020-09-21:
Silicon bug or bad driver code? And it has nothing to do with the ADC, I do not use it.

wreyford, 2021-01-21:
I ran into exactly the same issue. I've spent hours reading the documentation and trying to figure out why.
I'm using GPIO36 set up as an input pin for interrupt on falling edge (GPIO_INTR_NEGEDGE), to sense a Capacitive Touch Panel Interrupt, and was getting multiple interrupts being triggered while not touching. At first I thought it was my code.
Eventually I commented out all the other code, and **noticed that it worked properly when no Wifi was enabled**.
I pulled the IO36 high by wiring it to VCC, then ran the code without Wifi, and got no spurious interrupts, but as soon as Wifi was  enabled, I got spurious interrupts in spite of the IO36pin being pulled high by direct connection to Vcc. Perhaps this is a shielding problem with the antennae being so close to IO36, or Wifi is using this pin in an undocumented manner. I hope we get an answer to this problem from Espressif.

igrr, 2021-01-21:
Hi folks, sorry we forgot to update this issue when we closed https://github.com/espressif/esp-idf/issues/4117. The issue you are observing is a hardware bug, as mentioned in section 3.11 of https://espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf. The workaround for this issue has been recently merged into master branch in. A note has been added in the docs on how to enable this workaround in the application: https://github.com/espressif/esp-idf/commit/d890a516a1097f0a07788e203fdb1a82bb83520e.

mirko, 2021-01-21:
@igrr As mentioned by more than just me in this ticket, 3.11 doesn't cover the issue - at least not sufficiently.
It's *not* only happening when powering up peripherals and it's *not* only happening when using ADC.

stuartpittaway, 2021-03-06:
> And why on earth do GPIO36/39 only toggle states, when being pulled high? No interrupts triggered when left floating.

Agree - I'm also seeing this issue, no problem when floating, but GPIO36 triggers "fake" interrupts when pulled high and WIFI enabled (I'm not using the ADC or any power saving features)

HarmOtten, 2021-04-20:
Is the problem only the fact that interrupts keeps firing?
In that case a workaround could be: Polling GPIO36/39 in the loop.
Am I right?

mirko, 2022-12-12:
Why this ticket got closed and/or what's the relation to the issue mentioned here. Is it an acknowledged bug?
If in HW, is it (finally) documented?
If in SW, is it fixed - and if so, when (revision) and where (file / diff)?

C-Enjoyer, 2022-12-22:
Hi!
I'm facing similar problems with bluetooth. The moment i call `esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT)` interrupts start to fire constantly on pin 36 and 39.
They are both configured as external interrupts on the falling edge.

C-Enjoyer, 2022-12-22:
I figured out a workaround that does the job for me.
In my case the actual pins in hardware are working fine. Eventhough i get incorrect interrupts on the pins, the actual level doesn't change.
I noticed that it's possible to read the correct pin levels via `gpio_get_level()`.
If there is a real exti on the pins, an interrupts get's triggered correctly, too.

So on every interrupt i compare the pin level with the previous pin level and that's how i can figure out if a real interrupt happended.

Pseudo code would be something like that (a don't know how to write proper pseudo code):
```c
int pinLevels[GPIO_PIN_COUNT];

void init(void)
{
    configExti(); // configure all used pins as GPIO_INTR_ANYEDGE
    readPinLevels(); // read pin levels of all exti pins to "int pinLevels[GPIO_PIN_COUNT]"
}

void onExti(uint32_t pinNum)
{
    gpio_ex_extiEvent_t event;
	
    if (!getEvent(&event, pin))
    {
        return;
    }

    // use pin and event here
}

bool getEvent(gpio_ex_extiEvent_t* event, uint32_t pin)
{
    int level = gpio_get_level(pin);
    int prevLevel = pinLevels[pin];
	
    pinLevels[pin] = level;
	
    if (prevLevel == 0 && level == 1)
    {
        *event = GPIO_EX_EXTI_RISING;
        return true;
    }
    else if (prevLevel == 1 && level == 0)
    {
        *event = GPIO_EX_EXTI_FALLING;
        return true;
    }
	
    return false;
}
```

felixcollins, 2023-08-04:
I'm seeing this issue with phantom interrupts too. It was working fine then I updated to ADF v2.5-82-gbcad553 and IDF v4.4.4-698-gcc71308e2f and the issue appeared. Before that I was on IDF #3C86C2213C from 20 March 23, so in that time someone broke something in the GPIO interrupts....   The GPIO I'm having issues with is 39. I am running wifi and adc but I set a flag to ignore interrupts during adc operations and when enabling WIFI.