espressif/esp-idf #1097: how to handle two event handlers of wifi

jaggi, 2017-10-10:
we are working on sntp and  mqtt examples together, but we are getting evnt handler implicit declaration, so can someone how to use two different event in that handler function.
******************************************************************************************************

In file included from /home/user/esp32/esp-idf/esp-mqtt-master/oTA_WITH_sub_pub/main/./main.c:94:0:
/home/user/esp32/esp-idf/esp-mqtt-master/main/./mqtt_wifi.h:80:18: error: redefinition of 'event_handler'
 static esp_err_t event_handler(void *ctx, system_event_t *event) {
                  ^
In file included from /home/user/esp32/esp-idf/esp-mqtt-master/main/./main.c:93:0:
/home/user/esp32/esp-idf/esp-mqtt-master/oTA_WITH_sub_pub/main/./ota.h:45:18: note: previous definition of 'event_handler' was here
 static esp_err_t event_handler(void *ctx, system_event_t *event)
                  ^
make[1]: *** [main.o] Error 1
make: *** [main-build] Error 2

nkolban, 2017-10-10:
When you write an ESP-IDF application that uses WiFi, it is your responsibility to write an event handler which is invoked when important lifecycle events occur relating to WiFi.  If you have logic that you want executed when an event occurs, you must merge the execution of that logic into your handler.  If you have two sample applications that perform different work items when a WiFi event arrives, you must manually merge the code (or chain the handlers together) to achieve the task.  You can't register multiple handlers in ESP-IDF.

FayeY, 2017-10-13:
Hi @jaggi , did the above suggestion help you resolve the issue?

closed 2017-10-28 by FayeY

PranavD10, 2019-11-11:
I tried handling the WiFi events with two handlers, one handler where actual logic is written and the second with only the occurrence of the event is printed. It works in this way. I believe as long as you don't place the same logic in both handlers, you are good to go.