espressif/esp-idf #1099: build error：undefined reference to `wpa2_machine_start'

yuanfeng5721, 2017-10-10:
LINK build/application.elf
~/work/esp/esp-idf/components/esp32/lib/libnet80211.a(wl_cnx.o):(.text.cnx_start_handoff_cb+0x0): undefined reference to `wpa2_machine_start'

YanMinge, 2017-10-22:
I got the same issue

projectgus, 2017-10-23:
You may have out of date WiFi libraries. Run "git submodule update --init --recursive". Does the problem still occur?

YanMinge, 2017-10-23:
I'm sure I have done the command "git submodule update --init --recursive", and then I post the issue.
please try it.

projectgus, 2017-10-23:
> please try it

The builds work for us in general, and all of the examples are built by the CI system on every commit. So there is another factor here which is causing it to fail. Any information that you can provide is useful.

Can you please run these two commands:
```
mv build build_old
make
```
and see if the build succeeds the second time?

If it succeeds, then it is a bug in the build system. Can you put the build_old directory into a zip file and post it somewhere for us to look at?

YanMinge, 2017-10-23:
yes, I also do this work. and I also can build some sample code succeed. but if I use wifi like ftp and request,  it will failed here.

projectgus, 2017-10-24:
@YanMinge can you please run "make V=1" and copy the full build output to pastebin.com or gist.github.com ?

YanMinge, 2017-10-24:
Hi projectgus,

Thanks for your support. I have commit all my log here:

https://gist.github.com/YanMinge/098d1ed80d41f601405c8ed4b79fbf91

projectgus, 2017-10-24:
@YanMinge you seem to be building MicroPython not an ESP-IDF app. MicroPython has their own build system, so this is really an issue in micropython-esp32 not IDF.

The last line with the linker has the following libraries `-lcore -lnet80211 -lphy -lrtc -lpp -lwpa -lsmartconfig -lcoexist`. The library with `wpa2_machine_start` symbol is `libwpa2` so `-lwpa2` needs to be added to this list of libraries.

The line is here:
https://github.com/micropython/micropython-esp32/blob/esp32/ports/esp32/Makefile#L618

My guess is that you have updated ESP-IDF to a newer version than the version supported by MicroPython. The MicroPython build system should check the version here:
https://github.com/micropython/micropython-esp32/blob/esp32/ports/esp32/Makefile#L31
But in any case this is an issue with micropython-esp32 not ESP-IDF.

projectgus, 2017-10-24:
@yuanfeng5721 Can you please also check `git submodule update --init --recursive` to make sure you have newest submodules. If your build is still failing then please post the output of `make V=1` to pastebin.com or gist.github.com. Thanks.

YanMinge, 2017-10-24:
Thanks, I'm sure I have the newest submodules,  and I don't use the micropython's esp-idf. Because they are using a too old esp-idf. but I need use esp-idf that support external SRAM. I will try your advice and then quickly reply

projectgus, 2017-10-24:
> I don't use the micropython's esp-idf

This is why it doesn't build. It's not a bug in ESP-IDF or a bug in MicroPython, you've just made a custom mash-up of incompatible parts. Perhaps you can volunteer to help the MicroPython ESP32 project to make this combination work.

You may also be interested in this fork of MicroPython which already adds PSRAM and uses the IDF build system:
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo

YanMinge, 2017-10-24:
Yes, it is not a bug ,  and I have no experience to deal with esp-idf  *.a  library’s dependent library problem,  so I thought it was a library issue.  But now, I know how to slove such problem. 

We can build a sample code that need use the *.a library with command `make V=1` and then we can get the log just like this:

`esp-idf/components/esp32/lib -lcore -lrtc -lnet80211 -lpp -lwpa -lsmartconfig -lcoexist -lwps -lwpa2 -lespnow -lphy -L`

It contains the dependencies of the * .a library.

Thank you very much for your help！！！

closed 2017-11-22 by FayeY