espressif/esp-idf #1102: Where can I find "gethostname" and "gai_strerror"

Gfast2, 2017-10-11:
Hi ESP-IDF, 🙉 

I'm trying to port library ***open62541*** from _Linux/windows/QNX_ to _ESP-IDF_ with no luck ([***Issue***](https://github.com/espressif/esp-idf)). My new strategy is file for file to do this cross-compile.
The most problematic file is [**`ua_network_tcp.c`**](https://github.com/open62541/open62541/blob/0.2/plugins/ua_network_tcp.c). Durcing the compiling time for this file, There comes two errors:
```
/home/gfast2/workspace/ioc-110-ethernet/components/open62541/open62541/plugins/ua_network_tcp.c: In function 'ServerNetworkLayerTCP_start':
/home/gfast2/workspace/ioc-110-ethernet/components/open62541/open62541/plugins/ua_network_tcp.c:391:8: error: implicit declaration of function 'gethostname' [-Werror=implicit-function-declaration]
     if(gethostname(hostname, 255) == 0) {
        ^
/home/gfast2/workspace/ioc-110-ethernet/components/open62541/open62541/plugins/ua_network_tcp.c: In function 'UA_ClientConnectionTCP':
/home/gfast2/workspace/ioc-110-ethernet/components/open62541/open62541/plugins/ua_network_tcp.c:709:34: error: implicit declaration of function 'gai_strerror' [-Werror=implicit-function-declaration]
                        hostname, gai_strerror(error));
                                  ^

```

---

I `grep` all files from _ESP-IDF_. And found some examples (like `coap/libcoap/examples/coap-server.c`) implementate function `gai_strerror`. But nowhere defined this function (`gethostname` has the similar issue). What should I do? 🉑 

---

P.s:
I know my biggest problem right now is the part between my ears. But I did investigate a lot of time for this porting job. And I believe I've learned tons from this. 😃

negativekelvin, 2017-10-11:
https://github.com/espressif/esp-idf/blob/7d436c9a44b7506c3de4d5ed6decff8f35f64b67/components/tcpip_adapter/tcpip_adapter_lwip.c#L1024

Gfast2, 2017-10-11:
Hi @negativekelvin 
Thanks. This `gethostname` issue is firstly through the following statement addressed:

```
if(tcpip_adapter_get_hostname(TCPIP_ADAPTER_IF_STA, ((const char **)(hostname)) ) == 0) {
```

---

But I'd like to know something more about `gai_strerror`. It seems like as macro of `glibc` and served as for debugging perpose. 🐤

negativekelvin, 2017-10-11:
It just return string describing error code, you can just print the code instead.

Gfast2, 2017-10-12:
Thanks @negativekelvin , 🤘 🤝 
After fixing these problems, file **`ua_network_tcp.c`** get compiled beautifully. I believe I have to investigate some time to mangling the tcp connector in order to let ***OPC UA*** start to works on ESP32! 

---
BTW:
in **`ESP-IDF`** 's components folder. There are some examples from this components (like [**`~/esp/esp-idf/components/coap/libcoap/examples`**](https://github.com/obgm/libcoap/tree/6468887a12666f88b8704d797fc176cd4f40ee4c/examples)). What kind of examples are they? Are they ready to be used for **`ESP-IDF`** or only from this component's repo directly? 👂

closed 2017-10-12 by Gfast2