espressif/esp-idf #1103: Unable to get I2S working in Slave mode

pad52, 2017-10-11:
Trying with [ESP32_MP3_Decoder](https://github.com/MrBuddyCasino/ESP32_MP3_Decoder) from @MrBuddyCasino

Everything works as expected in master mode,
but if I try to set up Slave mode:
i2s_mode_t mode = I2S_MODE_SLAVE | I2S_MODE_TX;

Esp32 is correctly not generating clock but waiting for it and replying with data,
however the generated data is full of noise even if it's correctly synchronized.

costaud, 2017-10-18:
Would you please post your I2C initialisation code here, so that we can make some quick test?

pad52, 2017-10-18:
I just modified a couple of lines in [https://github.com/MrBuddyCasino/ESP32_MP3_Decoder](url)
to obtain the SLAVE MODE.

`static void init_i2s(renderer_config_t *config)
{
    i2s_mode_t mode = I2S_MODE_SLAVE | I2S_MODE_TX;
    i2s_comm_format_t comm_fmt = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB;

    [...]

    i2s_config_t i2s_config = {
            .mode = mode,          // Only TX
            .sample_rate = config->sample_rate, // 44100
            .bits_per_sample = config->bit_depth, // I2S_BITS_PER_SAMPLE_32BIT
            .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,   // 2-channels
            .communication_format = comm_fmt,
            .dma_buf_count = 64,                            // number of buffers, 128 max.
            .dma_buf_len = 128,                          // size of each buffer
            .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1        // Interrupt level 1
    };

    i2s_pin_config_t pin_config = {
            .bck_io_num = GPIO_NUM_26,
            .ws_io_num = GPIO_NUM_25,
            .data_out_num = GPIO_NUM_21,
            .data_in_num = I2S_PIN_NO_CHANGE
    };

    i2s_driver_install(config->i2s_num, &i2s_config, 1, &i2s_event_queue);

   [...]
        i2s_set_pin(config->i2s_num, &pin_config);
    [...]

    i2s_stop(config->i2s_num);
}
`

And this is the procedure to start i2s:
`
void renderer_start()
{
    if(renderer_status == RUNNING)
        return;

    renderer_status = RUNNING;
    i2s_start(renderer_instance->i2s_num);
	i2s_set_clk(renderer_instance->i2s_num, 44100, I2S_BITS_PER_SAMPLE_32BIT, I2S_CHANNEL_STEREO);

    // buffer might contain noise
    i2s_zero_dma_buffer(renderer_instance->i2s_num);

}`

And to push out samples:

`const char samp64[8] = {0, 0, ptr_l[0], ptr_l[1], 0, 0, ptr_r[0], ptr_r[1]};
                    bytes_pushed = i2s_push_sample(renderer_instance->i2s_num, (const char*) &samp64, max_wait);
                    break;
`

Anyway I get the same problem using the I2S example in esp-idf (_Before the last commit wich introduces APLL settings_ I need to try this new one!).
This is the i2s configuration:
`    i2s_config_t i2s_config = {
        .mode = I2S_MODE_SLAVE | I2S_MODE_TX,                                  // Only TX
        .sample_rate = SAMPLE_RATE,
        .bits_per_sample = 16,                                              
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,                           //2-channels
        .communication_format = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB,
        .dma_buf_count = 6,
        .dma_buf_len = 60,                                                      //
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1                                //Interrupt level 1
    };`

Thank you very much for your time

pad52, 2017-11-02:
With the new esp-idf version, using apll everything works. Thank you for your kind work!

closed 2017-11-02 by pad52