espressif/esp-idf #1105: GURU Meditation Error

rogei, 2017-10-11:
Hello,
the chip keeps rebooting all the time.

`
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:5544
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:12476
entry 0x40078f74
W (55) rtc_clk: Bogus XTAL frequency: 0 MHz
I (35) boot: ESP-IDF v3.0-dev-782-ge6afe28 2nd stage bootloader
I (36) boot: compile time 14:25:38
I (44) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (45) boot: SPI Mode       : DIO
I (49) boot: SPI Flash Size : 4MB
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x07ae4 ( 31460) map
I (111) esp_image: segment 1: paddr=0x00017b0c vaddr=0x3ffb0000 size=0x01ee0 (  7904) load
I (114) esp_image: segment 2: paddr=0x000199f4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/roge/esp/esp-idf/components/freertos/./xtensa_vectors.S:1675

I (118) esp_image: segment 3: paddr=0x00019dfc vaddr=0x40080400 size=0x06214 ( 25108) load
I (137) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x11650 ( 71248) map
0x400d0018: _stext at ??:?

I (162) esp_image: segment 5: paddr=0x00031670 vaddr=0x40086614 size=0x022bc (  8892) load
0x40086614: merge_adjacent at /home/roge/esp/esp-idf/components/heap/./multi_heap.c:194

I (166) esp_image: segment 6: paddr=0x00033934 vaddr=0x400c0000 size=0x00000 (     0) load
I (174) boot: Loaded app from partition at offset 0x10000
I (174) boot: Disabling RNG early entropy source...
I (180) cpu_start: Pro cpu up.
I (184) cpu_start: Starting app cpu, entry point is 0x40080fd4
0x40080fd4: call_start_cpu1 at /home/roge/esp/esp-idf/components/esp32/./cpu_start.c:219

I (166) cpu_start: App cpu up.
I (195) heap_init: Initializing. RAM available for dynamic allocation:
I (201) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (207) heap_init: At 3FFB26F0 len 0002D910 (182 KiB): DRAM
I (214) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (220) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (226) heap_init: At 400888D0 len 00017730 (93 KiB): IRAM
I (233) cpu_start: Pro cpu start user code
I (250) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error of type StoreProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400df33c  PS      : 0x00060730  A0      : 0x800dfdd8  A1      : 0x3ffb8f10  
0x400df33c: i2c_cmd_link_append at /home/roge/esp/esp-idf/components/driver/./i2c.c:821

A2      : 0x3ffaffd0  A3      : 0x3ffb8f40  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x800df33c  A9      : 0x3ffb8ef0  
A10     : 0x3ffaffd0  A11     : 0x400823cc  A12     : 0x00000000  A13     : 0x3ff67000  
0x400823cc: _malloc_r at /home/roge/esp/esp-idf/components/newlib/./syscalls.c:36

A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400df33c:0x3ffb8f10 0x400dfdd5:0x3ffb8f30 0x400d1f53:0x3ffb8f70 0x400d1f8b:0x3ffb8fb0 0x400d08c3:0x3ffb8fd0
0x400df33c: i2c_cmd_link_append at /home/roge/esp/esp-idf/components/driver/./i2c.c:821

0x400dfdd5: i2c_master_start at /home/roge/esp/esp-idf/components/driver/./i2c.c:821

0x400d1f53: i2c_sensor_init at /home/roge/workspace/Druck/main/./Druck_main.c:191

0x400d1f8b: app_main at /home/roge/workspace/Druck/main/./Druck_main.c:321

0x400d08c3: main_task at /home/roge/esp/esp-idf/components/esp32/./cpu_start.c:403

Rebooting...

`
make flash works perfectly.
In menuconfig reserve memory for 2 cores is enabled, and  "freertos runs on first core" is disabled.

When I disable all RTOS-functions the chip boots normally.

Are there any ideas what's wrong?

Thank You!

pad52, 2017-10-11:
I had that problem with I2S example using latest buildtool (crosstool-NG crosstool-ng-1.22.0-73-ge28a011) 5.2.0.
If I downgrade the buildtools using previous (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0
it fixes my problem - don't know why

rogei, 2017-10-12:
Thank you for your answer.
I tried this, but the previous buildtool doesn't work either...

igrr, 2017-11-02:
Hi @rogei, we have fixed a race condition in heap allocator in, b0c5665f15c8539fc0c9d1ebdb605e83dd6c79ce, i think it could make the malloc fail like this. Could you please update IDF and try again? Thanks.

closed 2017-11-14 by FayeY