espressif/esp-idf #1106: Issue: WPA2 EAP-GTC

Flyb29, 2017-10-11:
Hi,

i'm working on a project for my apprenticeship but can't connect my ESP32 to our WiFi. 
A colleague told me we're using WPA2 Enterprise with inner GTC (Generic Token Card) so I don't need any certificate.. Just username and password.

is it possible to connect the esp32 to WPA2 with GTC ? if its possible, how? 

thank you very much in advance!
flyb

XinDeng11, 2017-11-26:
@Flyb29, From your requirement, it seems that you don't need certificate verify, maybe you can try to mark esp_wifi_sta_set_wpa2_ent_set_ca_cert() and esp_wifi_sta_wpa2_ent_set_cert_key(), and have a try again. If it not working. maybe you can offer the sniffer log and then we will try to analysis

FayeY, 2018-02-03:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.

closed 2018-02-03 by FayeY