espressif/esp-idf #1108: ulp_adc current measurment is not correct

baoshi, 2017-10-12:
In the examples\system\ulp_adc
The claimed current draw @ 10Hz is 80uA
I found it is can be much less at round 39uA (below scope capture from uCurrent, 1mV=1uA)
[image]
This is possibly because GPIO15 has internal PU enabled but the pin is grounded to suppress boot message.
Add
`rtc_gpio_pullup_dis(GPIO_NUM_15);`
at beginning of the code resolves this issue.

igrr, 2017-10-12:
Just to clarify, do you get 39uA with or without this extra rtc_gpio_pullup_dis line?

baoshi, 2017-10-12:
Without rtc_gpio_pullup_dis, with GPIO15 tired to gnd-> 80uA
With rtc_gpio_pullup_dis, without GPIO15 tired to gnd -> 39uA
With/without rtc_gpio_pullup_dis, GPIO15 floating -> 39uA

negativekelvin, 2017-10-12:
Avg current with spikes probably at least +10ua?

closed 2017-11-13 by igrr