espressif/esp-idf #1110: APLL / GPIO16_EMAC_CLK_OUT - wrong frequency - unable to generate 50MHz clk with rev0 chip

sauttefk, 2017-10-12:
I want to use the APLL to generate the required 50MHz for an external Ethernet PHY using the APLL and the EMAC_CLK_OUT on GPIO16.

### Somehow the output frequency is unexpectedly divided by 5

According to  the [_ECO and Workarounds for Bugs in ESP32_ (issue 3.7)](https://espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf) document the formula for the APLL frequency for rev0 chips is:
`f_out = f_xtal × (sdm2 + 4) / (2 × (o_div + 2))` where `f_xtal` is **40MHz**, `sdm2` is **6** (4th parameter), o_div is **2** (5th parameter).
This evaluates to `f_out = 40MHz × (6 + 4) / (2 × (2 + 2))` =>` f_out = 400MHz / 8` => `f_out = 50MHz`
According to the [_ESP32 Technical Reference Manual_ (paragraph 3.2.7)](http://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) the dividend of the formula (`f_xtal × (sdm2 + 4)`) has a frequency range from 350MHz to 500MHz (Where 400MHz lies well within).

I'm setting up set the correct parameters with `rtc_clk_apll_enable(1, 0, 0, 6, 2)` but the frequency I get on `GPIO16` is **10MHz** instead of **50MHz**.

When I change `o_div` from **2** to **0**, I get **20MHz** (which prooves that the divisor works as described).
When I change `sdm2` from **6** to **8**, I get **12MHz** (which prooves that the dividend works as described in the formula).
When I change `sdm2` to **1** and `o_div` to **0** (`rtc_clk_apll_enable(1, 0, 0, 1, 0)`) - this also evaluates to 50MHz - the function call never returns, as the dividend now gets 200MHz which is out of the frequency range of the APLL but rtc_clk_apll_enable waits for a PLL-lock which never happens (which prooves that the dividend is lower than the specified minimum).
When I change `sdm2` to **11** and `o_div` to **4**  (`rtc_clk_apll_enable(1, 0, 0, 11, 4)`) - this also evaluates to 50MHz - but the dividend is now 600MHz and the PLL gets no lock (which prooves that the dividend is higher than the specified maximum).

**My questions:**
* Is there any by 5 divider which has to be disabled?
* Is the APLL somehow using the internal 8MHz clock instead of the external 40MHz crystal?
* Can anybody provide me some code that generates a 50MHz clk on GPIO16?

The code that I'm currently using is as follows:
```
#include "soc/rtc.h"
#include "soc/emac_ex_reg.h"

ESP_LOGD(TAG, "################ set up EMAC_CLK_OUT #############");
PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO16_U, FUNC_GPIO16_EMAC_CLK_OUT);
REG_SET_FIELD(EMAC_EX_CLKOUT_CONF_REG, EMAC_EX_CLK_OUT_H_DIV_NUM, 0);
REG_SET_FIELD(EMAC_EX_CLKOUT_CONF_REG, EMAC_EX_CLK_OUT_DIV_NUM, 0);
REG_CLR_BIT(EMAC_EX_CLK_CTRL_REG, EMAC_EX_EXT_OSC_EN);
REG_SET_BIT(EMAC_EX_CLK_CTRL_REG, EMAC_EX_INT_OSC_EN);
ESP_LOGD(TAG, "################ before enabling APLL ############");
rtc_clk_apll_enable(1, 0, 0, 6, 2); // 50 MHz
ESP_LOGD(TAG, "################ APLL has a stable lock ##########");

```

sauttefk, 2017-10-12:
@igrrIs there any further information (register description, functional schematic) available for the APLL module within the ESP32?

sauttefk, 2017-10-12:
@igrr 
Is there any further information available (register description, functional schematic) for the APLL module within the ESP32, that you could make public available?
The only thing we have is the .H file  `components/soc/esp32/i2c_apll.h` but this leaves a lot of question marks on all the registers that are not set up in `components/soc/esp32/rtc_clk.c` and` rtc_clk_apll_enable`

igrr, 2017-10-12:
They way you configure APLL is correct, i see no issue there.

I have copied above code into an empty application, added `periph_module_enable(PERIPH_EMAC_MODULE);`, and observed the waveform on GPIO16 — its frequency is 50MHz, as expected. Can you try doing that as well?

Perhaps there is some other part of your application which may be affecting this?

For debugging purposes, you may also observe APLL output on GPIO0 like this:
```c++
    rtc_clk_apll_enable(true, 0, 0, 6, 2);
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
    REG_WRITE(PIN_CTRL, 6);
```

sauttefk, 2017-10-12:
@igrr thanks for your quick answer!
I still get 10MHz on GPIO16, **but also 50MHz on GPIO0** thanks to the hint you gave me.
I'm now using GPIO0 as clock source for the LAN8720 PHY. It gets an IP-address and is pinging 👍 .
Using GPIO0 as clock output is even better for my design where all GPIOs are used up as this does not interfere when bootstrapping the ESP32.

I create a pull request for this setup with new menuconfig options for APLL clock generation for Ethernet PHYs.

igrr, 2017-10-13:
Sending ethernet clock output via GPIO0 is somewhat of a hack... it would be nice to figure out why the original way did not work. Would you be able to share the complete code (which also initialized Ethernet driver, i assume)?

sauttefk, 2017-10-13:
@igrr
its basically the ethernet example with some minor modifications https://github.com/espressif/esp-idf/tree/master/examples/ethernet/ethernet
How should I send you the modifications I've done?

sauttefk, 2017-10-13:
@igrr 
Sure its a hack, but using GPIO0 is even better than GPIO16, as GPIO0 becomes an output after the bootloader finished and the state of the clock signal is irrelevant for the Ethernet PHY. => No special care for GPIO0 during bootstrapping, one more freed GPIO (GPIO16).

igrr, 2017-10-13:
I think that clock source should be configurable when ethernet driver is initialized.

If you can point me to that repository containing your changes (which have the above mentioned issue with output clock frequency), that would be much appreciated. We will then try to debug why the clock signal is getting divided. Once that is done, Ethernet driver can be updated to include this new option (use APLL or external clock).

sauttefk, 2017-10-13:
[image]
This is a screenshot from my scope when both outputs are enabled.
`GPIO16` **10MHz**
`GPIO0` **50MHz**
This is from a revision 1 chip.

igrr, 2017-10-13:
Could you please upload the full code you used to get this result?

sauttefk, 2017-10-15:
OK, I found the solution...

There is an additional divisor stage before the clk-signal is put out on GPIO16.
Until the bit `DPORT_WIFI_CLK_EMAC_EN` is set, `EMAC_EX_CLK_OUT_H_DIV_NUM` is constantly reset to `0x2` and `EMAC_EX_CLK_OUT_DIV_NUM` is defaulting to `0x4`.
`DPORT_WIFI_CLK_EMAC_EN` is set in `periph_module_enable(PERIPH_EMAC_MODULE);`

To get 50MHz on GPIO16 you have to `periph_module_enable(PERIPH_EMAC_MODULE)` before you set up the `CLK_OUT_DIV`. Just like in the following code fragment:
```
  periph_module_enable(PERIPH_EMAC_MODULE);
  // GPIO16_EMAC_CLK_OUT divisor - 
  //   default: EMAC_EX_CLK_OUT_H_DIV_NUM: 0x2 
  //   default: EMAC_EX_CLK_OUT_DIV_NUM:   0x4
  PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO16_U, FUNC_GPIO16_EMAC_CLK_OUT);
  REG_SET_FIELD(EMAC_EX_CLKOUT_CONF_REG, EMAC_EX_CLK_OUT_H_DIV_NUM, 0);
  REG_SET_FIELD(EMAC_EX_CLKOUT_CONF_REG, EMAC_EX_CLK_OUT_DIV_NUM, 0);
  emac_enable_clk(true);
  REG_SET_FIELD(EMAC_EX_PHYINF_CONF_REG, EMAC_EX_PHY_INTF_SEL, EMAC_EX_PHY_INTF_RMII);
  emac_dma_init();
  // rmii internal clock
  REG_CLR_BIT(EMAC_EX_CLK_CTRL_REG, EMAC_EX_EXT_OSC_EN);
  REG_SET_BIT(EMAC_EX_CLK_CTRL_REG, EMAC_EX_INT_OSC_EN);
```

FayeY, 2017-10-16:
Is this issue solved? Can we close it?

sauttefk, 2017-10-16:
I will make a pull request with some new code to make those clock signals available for the PHY

closed 2017-10-16 by sauttefk

cnlohr, 2018-07-10:
Just wondering, was anyone able to get an inverted (or heavily phase shifted) clock out of GPIO0? It seems it's difficult to get the phase and shift right for something like the LAN8720.  I've tried a number of clock settings and none seem to get what I want.

mrdc, 2021-09-28:
> Sending ethernet clock output via GPIO0 is somewhat of a hack..

@igrr 

Can I use this approach in hardware design or it's not recommended? I mean GPIO_0 as output for ESP32 Wrover.