espressif/esp-idf #1111: How to specify compiler standard

Gfast2, 2017-10-12:
Hallo ESP-IDF,

As you know, I'm working on porting [open62541](https://open62541.org/) to here. I'd like to know how tune compiler standard?

As far as I can understand. The default **`gnu99`** is the default one, **`-Werror=all`** is turned on.

Thanks! 💃

igrr, 2017-10-12:
Extra compiler flags can be specified for each component, by setting CFLAGS
in the component makefile. Please see
http://esp-idf.readthedocs.io/en/latest/api-guides/build-system.html#optional-component-specific-variables

Gfast2, 2017-10-12:
Hi @igrr ,

Thanks. I've  tried to define this in this component's `component.mk` file:
```
CFLAGS := -Og -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DWITH_POSIX -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H  -g   -pipe -Wextra -Wformat -Wno-unused-parameter -Wno-unused-function -Wno-unused-label -Wpointer-arith -Wreturn-type -Wsign-compare -Wmultichar -Wstrict-overflow -Wcast-qual -Wmissing-prototypes -Winit-self -Wuninitialized -Wformat-security -Wformat-nonliteral -Wshadow -Wconversion -fvisibility=hidden -fPIC
```
But where should I call this variable in this file? ❔ 

My goal is to suppress the compiler treat all ***warning*** as ***errors*** and do not specify the ***C standard***. Because I'm trying to deploy `cmake` in ESP-IDF to build [**this**](https://open62541.org/) libray.

This is how my [**component.mk**](https://gist.github.com/Gfast2/5814f3fbcedb1493bfe7f390ccd2ded5) for this component looks like.

**
The porting job is discussed in [***this issue***](https://github.com/espressif/esp-idf/issues/921)

Gfast2, 2017-10-12:
I've updated the **`component.mk`** mentioned above. 
Now the compiler let me pass the _warnings_ firstly.
The lessons I've learned is: for this perpose, I need to specify what I'd like the _C Preprocessor_ do. I don't need to called its variable (***CPPFLAGS***) anywhere. _**kconfig**_ will read these settings according to the specified name.
Thanks again. 😀

closed 2017-10-12 by Gfast2