espressif/esp-idf #1112: esptool.exe missing

bhaessig83, 2017-10-12:
Hi everybody,

I tried today ton install the Arduino core for the ESP32 but I have some problems,
When I ran the get.exe in the Documents\Arduino\hardware\espressif\esp32\tools folder, I got a "SSL: CERTIFICATE_VERIFY_FAILED" error as shown below.

[image]

I looked at the issues and extract the xtensa-esp32-elf-win32-1.22.0-73-ge28a011-5.2.0.zip files in Documents\Arduino\hardware\espressif\esp32\tools. 

It solves the first step of the compilation problem, but now I get the following message in Arduino:

> Archiving built core (caching) in: C:\Users\bhaessig\AppData\Local\Temp\arduino_cache_441373\core\core_espressif_esp32_esp32-gateway_FlashFreq_80,UploadSpeed_115200_78ee4bc9f9bc86092d556a84099a7a06.a
> exec: "C:\\Users\\bhaessig\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/esptool.exe": file does not exist
> Erreur de compilation pour la carte OLIMEX ESP32-GATEWAY

Does anybody has an idea of the problem and could help me to solve it.
Thanks in advance for your help.

me-no-dev, 2017-10-12:
Why not post in the arduino repository? :)
It is unknown to me why the download would fail with that error, but you can get the archive from: https://dl.espressif.com/dl/esptool-4dab24e-windows.zip
put it in tools/dist folder and run get.exe again. If you need further help, please post in the Arduino repository for ESP32 :) Thanks

bhaessig83, 2017-10-12:
Sorry for the mistake,, I'm quite new in the Github world.
I've made a copy of esptool.exe in the directory. The get.exe doesn't work but the compilation with Arduino seems ok.

Thanks a lot

igrr, 2017-10-12:
We were moving dl.espressif.com to a new server today, so SSL: CERTIFICATE_VERIFY_FAILED might be related to that. If you remove all the downloaded files and try running get.exe again, it should work, i think. Closing this issue, if you get any more trouble with the installation, feel free to report on github.com/espressif/arduino-esp32. Thanks.

closed 2017-10-12 by igrr