espressif/esp-idf #1113: Feature request: pass custom parameters to callbacks

xarrange, 2017-10-12:
When all bugs are fixed, It would be nice to have option to pass cusom pointer to callbacks like in FreeRTOS tasks. It costs almost nothing.

e.g. what we have now in **esp_gatts_api**:

```
typedef void (* esp_gatts_cb_t)(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param);

esp_err_t esp_ble_gatts_register_callback(esp_gatts_cb_t callback);
```

I suggest:

```
typedef void (* esp_gatts_cb_t)(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param, void *cb_param);

esp_err_t esp_ble_gatts_register_callback(esp_gatts_cb_t callback, void *cb_param);
```

I did it locally and it helps a lot in c++ to reference object from static method.
And it's no only in BLE there are lots of such places all over IDF

Yulong-espressif, 2017-10-16:
@xarrange thanks for your suggest, what is the "void *cb_param" for?

xarrange, 2017-10-19:
for various purposes.
And as I mentioned I use it for c++:

```
class BLE
{
	public:
		BLE();
		void gatt_server_callback(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param);
		static void gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param, void *cb_param);
};

BLE::BLE(uint32_t id){
	esp_ble_gatts_register_callback(&BLE::gatts_event_handler, this);
}

void BLE::gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param, void *cb_param)
{	
	((BLE*)(cb_param))->gatt_server_callback(event, gatts_if, param);
}

void BLE::gatt_server_callback(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param) 
{
	switch (event) {
	...
	}
}
```

Yulong-espressif, 2017-11-04:
@xarrange Thanks for your suggestion.  Each parameter of the callback function is designed according to protocol and application requirements. All these parameters are common. If you want to add a parameter, you can add an adaptation layer to our API.

xarrange, 2017-11-06:
Can you give a brief example how to "add an adaptation layer"? Because I don't understand that.
What I'm proposing doesn't change protocols at all, it just add place for customization

FayeY, 2018-01-11:
Hi, so sorry that we are not going to change our APIs, please maintain customer parameters on your application. Sorry for the inconvenience. Thanks.

closed 2018-01-11 by FayeY

c0ffee, 2023-03-18:
hmm sigh, seems like the responders didn't understand the main issue here and how poorly this is designed for usage in c++