espressif/esp-idf #1114: Feature Request: Bluetooth RFCOMM/SPP as virtual uart in VFS

dmcnaugh, 2017-10-13:
**Is it on the ESP-IDF roadmap to provide a virtual uart integrated into the VFS that utilises RFCOMM/SPP over Bluetooth?**

Looking at the example code under
```
esp-idf/examples/bluetooth/ble_spp_server/
```
I see a lot of boilerplate and complexity that I would prefer to avoid.

mywang-espressif, 2017-10-30:
Hi @dmcnaugh , SPP for classic Bluetooth is under development, and we will deliver it in a month or a bit longer. The APIs are not decided yet and I don't know what they will be like. But we will try to make the APIs easy to use.

mringwal, 2017-11-21:
@dmcnaugh BTstack has full support for SPP. BTstack doesn't require a file system, so RFCOMM has a regular API. A simple SPP Server looks like this: https://github.com/bluekitchen/btstack/blob/master/example/spp_counter.c

Getting started with BTstack on ESP32: https://github.com/bluekitchen/btstack/tree/master/port/esp32

closed 2017-12-01 by FayeY