espressif/esp-idf #1115: Save Exception, Error or Core Dump to Flash

256dpi, 2017-10-13:
Is it possible to write exceptions, errors or core dumps to flash before rebooting device? That way one could read out the data on the next boot and send them off to a server for inspection and notification.

Do you already support something like this or are there any plans on integrating such a feature?

256dpi, 2017-10-13:
I just realized that it is already possible to save the Core Dump to flash and read it out later.

Now I'm wondering: Is the "saving core dump to flash" approach the recommended way of handling possible errors in a production application?

igrr, 2017-12-08:
Hi @256dpi, we do not recommend to keep core dump enabled in production applications, as it has the potential to leak secrets present on the stack of some task. Core dump currently does not support flash encryption, it always writes the data to flash unencrypted. It is usually okay to enable this feature for trial and pre-production runs though, depending on security considerations.

closed 2017-12-08 by igrr