espressif/esp-idf #1116: How to link a .a file which is compiled by cmake during kconfig compile the whole project

Gfast2, 2017-10-13:
Hi ESP-IDF,

I'm trying to port [***open62541***](https://open62541.org/) to ***ESP-IDF*** these days. The original Issue is [***this one***](https://github.com/espressif/esp-idf/issues/921). 🍰 

With the great help from you. I got the library compiled from its project source into file "open62541.a", which can be found in ***`[projectFolder]/build/open62541/`***. but now because of some setting issues in ***`kconfig`*** my compiler don't now where can it find this file.
```
xtensa-esp32-elf-gcc: error: libopen62541.a: No such file or directory
/home/gfast2/esp/esp-idf/make/project.mk:359: recipe for target '/home/gfast2/workspace/ioc-110-ethernet/build/ethernet_demo.elf' failed
make: *** [/home/gfast2/workspace/ioc-110-ethernet/build/ethernet_demo.elf] Error 1

```

The _VERBOSE_ log from the linker is:
```
xtensa-esp32-elf-gcc -nostdlib -u call_user_start_cpu0  -Wl,--gc-sections -Wl,-static -Wl,--start-group  -L/home/gfast2/workspace/ioc-110-ethernet/build/app_trace -lapp_trace -L/home/gfast2/workspace/ioc-110-ethernet/build/app_update -lapp_update -L/home/gfast2/workspace/ioc-110-ethernet/build/aws_iot  -L/home/gfast2/workspace/ioc-110-ethernet/build/bootloader_support -lbootloader_support -L/home/gfast2/workspace/ioc-110-ethernet/build/bt -lbt -L/home/gfast2/workspace/ioc-110-ethernet/build/coap -lcoap -L/home/gfast2/workspace/ioc-110-ethernet/build/console -lconsole -L/home/gfast2/workspace/ioc-110-ethernet/build/cxx -lcxx -u __cxa_guard_dummy -L/home/gfast2/workspace/ioc-110-ethernet/build/driver -ldriver -L/home/gfast2/workspace/ioc-110-ethernet/build/esp32 -lesp32 /home/gfast2/esp/esp-idf/components/esp32/libhal.a -L/home/gfast2/esp/esp-idf/components/esp32/lib -lcore -lrtc -lnet80211 -lpp -lwpa -lsmartconfig -lcoexist -lwps -lwpa2 -lespnow -lphy -L /home/gfast2/esp/esp-idf/components/esp32/ld -T esp32_out.ld -u ld_include_panic_highint_hdl -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.spiram_incompatible_fns.ld -L/home/gfast2/workspace/ioc-110-ethernet/build/esp_adc_cal -lesp_adc_cal -L/home/gfast2/workspace/ioc-110-ethernet/build/ethernet -lethernet -L/home/gfast2/workspace/ioc-110-ethernet/build/expat -lexpat -L/home/gfast2/workspace/ioc-110-ethernet/build/fatfs -lfatfs -L/home/gfast2/workspace/ioc-110-ethernet/build/freertos -lfreertos -Wl,--undefined=uxTopUsedPriority -L/home/gfast2/workspace/ioc-110-ethernet/build/heap -lheap -L/home/gfast2/workspace/ioc-110-ethernet/build/jsmn -ljsmn -L/home/gfast2/workspace/ioc-110-ethernet/build/json -ljson -L/home/gfast2/workspace/ioc-110-ethernet/build/libsodium -llibsodium -L/home/gfast2/workspace/ioc-110-ethernet/build/log -llog -L/home/gfast2/workspace/ioc-110-ethernet/build/lwip -llwip -L/home/gfast2/workspace/ioc-110-ethernet/build/main -lmain -L/home/gfast2/workspace/ioc-110-ethernet/build/mbedtls -lmbedtls -L/home/gfast2/workspace/ioc-110-ethernet/build/mdns -lmdns -L/home/gfast2/workspace/ioc-110-ethernet/build/micro-ecc -lmicro-ecc -L/home/gfast2/workspace/ioc-110-ethernet/build/mongoose -lmongoose -L/home/gfast2/workspace/ioc-110-ethernet/build/newlib /home/gfast2/esp/esp-idf/components/newlib/lib/libc.a /home/gfast2/esp/esp-idf/components/newlib/lib/libm.a -lnewlib -L/home/gfast2/workspace/ioc-110-ethernet/build/nghttp -lnghttp -L/home/gfast2/workspace/ioc-110-ethernet/build/nvs_flash -lnvs_flash -L/home/gfast2/workspace/ioc-110-ethernet/build/open62541 libopen62541.a -L/home/gfast2/workspace/ioc-110-ethernet/build/openssl -lopenssl -L/home/gfast2/workspace/ioc-110-ethernet/build/pthread -lpthread -L/home/gfast2/workspace/ioc-110-ethernet/build/sdmmc -lsdmmc -L/home/gfast2/workspace/ioc-110-ethernet/build/soc -lsoc -L/home/gfast2/workspace/ioc-110-ethernet/build/spi_flash -lspi_flash -L/home/gfast2/workspace/ioc-110-ethernet/build/spiffs -lspiffs -L/home/gfast2/workspace/ioc-110-ethernet/build/tcpip_adapter -ltcpip_adapter -L/home/gfast2/workspace/ioc-110-ethernet/build/ulp -lulp -L/home/gfast2/workspace/ioc-110-ethernet/build/vfs -lvfs -L/home/gfast2/workspace/ioc-110-ethernet/build/wear_levelling -lwear_levelling -L/home/gfast2/workspace/ioc-110-ethernet/build/wpa_supplicant -lwpa_supplicant -L/home/gfast2/workspace/ioc-110-ethernet/build/xtensa-debug-module -lxtensa-debug-module -lgcc -lstdc++ -lgcov -Wl,--end-group -Wl,-EL -o /home/gfast2/workspace/ioc-110-ethernet/build/ethernet_demo.elf -Wl,-Map=/home/gfast2/workspace/ioc-110-ethernet/build/ethernet_demo.map
```

I believe the prefix for libopen62541.a is not appended as it should. What can I do? 📿

Gfast2, 2017-10-13:
After I remove line [**`LIBRARY := libopen62541.a`**](https://gist.github.com/Gfast2/5814f3fbcedb1493bfe7f390ccd2ded5#file-component-mk-L5) from its `component.mk`, I get this library compiled without error.

---

But I'd like to know what cause this issue? What was wrong for my [`component.mk`](https://gist.github.com/Gfast2/5814f3fbcedb1493bfe7f390ccd2ded5)?

_P.s.
This issue will be leave open till end of this weekend. If there is no one wanna address it. I will close it on the up comming monday._

Gfast2, 2017-10-17:
If so many days no one is interested in answer this _issue_. Should I close it, or wait?

projectgus, 2017-10-17:
@Gfast2 , do you have a copy of your component.mk that we can refer to?

projectgus, 2017-10-17:
If you're using the exact one posted in the other issue, you are correct that the root issue is probably the path to LIBRARY. Maybe try replacing COMPONENT_ADD_LDFLAGS with something like:
```
COMPONENT_ADD_LDFLAGS := -L$(COMPONENT_BUILD_DIR)/ -lopen62541
```
Update the `-L` argument with the path where libopen62541.a is being built by the build step ( `$(COMPONENT_BUILD_DIR)` expands to `build/componentname`

In general, this kind of question is probably better suited to https://esp32.com with its discussion forum format.

Gfast2, 2017-10-17:
@projectgus ,

Thanks for answering my issue. I'll give it a try later. Since this question block my working progress no more. I'd like to close this. (in order to leave places for other developers's question) 🕚 

The `component.mk` that is related for this issue is [**`here***](https://gist.github.com/Gfast2/5814f3fbcedb1493bfe7f390ccd2ded5) (In case someone wanna check this out later)

---

I've tried to ask questions in https//esp32.com. But it seems like that the discussions here are more _vivid_. But I will do try to asking more questions there.

closed 2017-10-17 by Gfast2

Pratikhyadav, 2017-11-11:
Hello,

I want to port mxml library in esp-idf where i want to use my static libmxml.a. For that i added One folder in Component , it's structure is as following:

-- component
    -- libxml
       -- include
          (contain header file)
       -- lib
           (contain lib file libmxml.a)
       --component.mk
          COMPONENT_ADD_LDFLAGS=$(COMPONENT_PATH)/lib/libmxml.a -l$(COMPONENT_NAME)
          include $(IDF_PATH)/make/component_common.mk

--main
   --main.c
     contain application code 
  -- component.mk
    include $(IDF_PATH)/make/component_common.mk

after compiling this project i get following error:

/esp-idf-v2.1/components/libmxml/lib/libmxml.a: error adding symbols: File format not recognized

Am i Doing anything wrong?
Can anyone provide suggestion regarding this or guidance to port   library in esp-idf?

Gfast2, 2017-11-11:
Hi @Pratikhyadav ,

[***In my practices***](https://github.com/espressif/esp-idf/issues/921), some one has shown me how to config _`component.mk`_ to let _**ESP-IDF**_ deploy for example _`cmake`_ to compile a specific component. I believe they are somehow the same scenerio for you.

Please check out [***this Gist***](https://gist.github.com/Gfast2/5814f3fbcedb1493bfe7f390ccd2ded5) for inspiration. On that time, I think I specified the name of the compiled library in _`component.mk`_. Then the compiler told me she can't found that compiled library with the name I specified. After I comment out this line of code in the _`component.mk`_, Everything works out automatically right.

Please check out the revision of that Gist for detail. Here is the change:

```
 #Path and name of the generated library
-LIBRARY := libopen62541.a
+# LIBRARY := libopen62541.a
```

Pratikhyadav, 2017-11-11:
Thanks @Gfast2  For quick Reply,

I Think in your case you generating ".a" file every time when you compile your project. but I don't want to compile my library's source code. i just want to use that static library which i i have already generated.

I don't know about cmake. can you give me any reference to use cmake to generate component.mk and what changes required in library's make file (which will generate lib i.e. "mylib.a")? 
It would be great help.

Gfast2, 2017-11-11:
Please click the hot links in my last answers, which has that file as a Gist. Cheers

projectgus, 2017-11-15:
Hi @Pratikhyadav 

> after compiling this project i get following error:
>
> /esp-idf-v2.1/components/libmxml/lib/libmxml.a: error adding symbols: File format not recognized

Was the library file compiled and archived using the esp32 cross-toolchain?

> COMPONENT_ADD_LDFLAGS=$(COMPONENT_PATH)/lib/libmxml.a -l$(COMPONENT_NAME)

You don't need both these arguments. You should be able to either do:

```
> COMPONENT_ADD_LDFLAGS=$(COMPONENT_PATH)/lib/libmxml.a
```
OR
```
> COMPONENT_ADD_LDFLAGS= -L $(COMPONENT_PATH)/lib/libmxml.a -lmxml
```

In the second case, your component will need to be named something other than mxml as the build system will also generate an (empty) libmxml.a file in the build directory.